chipsalliance/verible #348: Formatter: Indent parameter and port list with two spaces

imphil, 2020-07-03:
The style guide used in OpenTitan and Ibex requires two space indent for parameter and port lists; the formatter seems to prefer four spaces.

Docs:
* https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#basic-template
* https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#module-declaration

Actual

```systemverilog
module something #(
    parameter int Width = 1600
) (
    input clk
);
```

Expected

```systemverilog
module something #(
  parameter int Width = 1600
) (
  input clk
);
```

fangism, 2020-07-03:
Duplicate of #40

imphil, 2020-07-03:
I thought there must have been an issue for it, but didn't find it. Sorry, let's continue the discussion in #40.

closed 2020-07-03 by imphil