chipsalliance/verible #350: Formatter: Align comments in port list

imphil, 2020-07-03:
The formatter lines up data types and port names, but doesn't line up comments. This would give slightly easier-to-read code.

https://github.com/lowRISC/ibex/blob/master/rtl/ibex_controller.sv

Expected:

```systemverilog
module something(
    // decoder related signals
    input logic illegal_insn_i,    // decoder has an invalid instr
    input logic ecall_insn_i,      // decoder has ECALL instr
    input logic mret_insn_i,       // decoder has MRET instr
    input logic dret_insn_i,       // decoder has DRET instr
    input logic wfi_insn_i,        // decoder has WFI instr
    input logic ebrk_insn_i,       // decoder has EBREAK instr
    input logic csr_pipe_flush_i,  // do CSR-related pipeline flush
);
endmodule
```

Actual:

```systemverilog
module something(
    // decoder related signals
    input logic illegal_insn_i,  // decoder has an invalid instr
    input logic ecall_insn_i,  // decoder has ECALL instr
    input logic mret_insn_i,  // decoder has MRET instr
    input logic dret_insn_i,  // decoder has DRET instr
    input logic wfi_insn_i,  // decoder has WFI instr
    input logic ebrk_insn_i,  // decoder has EBREAK instr
    input logic csr_pipe_flush_i,  // do CSR-related pipeline flush
);
endmodule
```

fangism, 2020-07-03:
b/158030789

fangism, 2020-07-03:
This is easy to do with the current column detection interface, however, aggressively aligning another column today increases the risk of column limit overflow. Today, the back-off handling (in the event of overflow) automatically disables alignment in such cases. We should consider a progressive back-off of alignment in parallel with this feature, to reduce the risk of missing out on alignment opportunities.

PR #613 (2021-05-18): Align comments in port list
Fixes #350
closed 2021-05-18 by hzeller