chipsalliance/verible #352: Formatter: Indent body of initializer lists

imphil, 2020-07-03:
STR: Run `vendor/lowrisc_ip/prim/rtl/prim_keccak.sv` (https://github.com/lowRISC/ibex/blob/master/vendor/lowrisc_ip/prim/rtl/prim_keccak.sv) through verible formatter.

Expected output, according to the lowRISC/ibex/opentitan style guide:
```systemverilog
localparam logic [63:0] RC[24] = '{
    64'h 0000_0000_0000_0001, // Round 0
    64'h 0000_0000_0000_8082, // Round 1
    64'h 8000_0000_8000_8008 // Round 23
};
```

Actual output:
```systemverilog
localparam logic [63:0] RC[24] = '{64'h0000_0000_0000_0001,  // Round 0
64'h0000_0000_0000_8082,  // Round 1
64'h8000_0000_8000_8008  // Round 23
};
```

Actually, the https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#line-wrapping isn't very definite on this topic beyond the need for an indent. In line with what Verible typically produces, a four-space indent seems the most appropriate choice.

On the location of the closing `};`: The lowRISC style guide shows it in the last line in the example, but doesn't explicitly mention that. I opened https://github.com/lowRISC/style-guides/issues/39 to see if there's clear guidance from the style guide. (But let's keep this bug on the indentation for now.)

fangism, 2020-07-03:
There are two related issues afoot here, one is the partitioning of the RHS expression, and the other is the indentation thereof.

hzeller, 2021-04-26:
Re-opened for now as the fix introduced a regression so disabled for now.

PR #755 (2021-04-14): Indent initializer lists
Fixes #352.
commit 9d454971 (2021-04-14): Indent initializer lists (PR #755)

Indent initializer lists
Fixes #352
PR #815 (2021-06-21): Fix initializer list indentation
Second take to fix #352.
Reworked partition creation, added test cases for multiple nested lists.
closed 2021-06-21 by tgorochowik