chipsalliance/verible #357: Formatter: Alignment of assignment expressions

GregAC, 2020-07-08:
The formatter doesn't align assignment expressions, here's a simple example from https://github.com/lowRISC/ibex/blob/master/rtl/ibex_wb_stage.sv:

Before formatting:
```systemverilog
always_ff @(posedge clk_i) begin
  if(en_wb_i) begin
    rf_we_wb_q      <= rf_we_id_i;
    rf_waddr_wb_q   <= rf_waddr_id_i;
    rf_wdata_wb_q   <= rf_wdata_id_i;
    wb_instr_type_q <= instr_type_wb_i;
    wb_pc_q         <= pc_id_i;
  end
end

assign rf_waddr_wb_o         = rf_waddr_wb_q;
assign rf_wdata_wb_mux[0]    = rf_wdata_wb_q;
assign rf_wdata_wb_mux_we[0] = rf_we_wb_q & wb_valid_q;
```

After formatting:
```systemverilog
always_ff @(posedge clk_i) begin
  if (en_wb_i) begin
    rf_we_wb_q <= rf_we_id_i;
    rf_waddr_wb_q <= rf_waddr_id_i;
    rf_wdata_wb_q <= rf_wdata_id_i;
    wb_instr_type_q <= instr_type_wb_i;
    wb_pc_q <= pc_id_i;
  end
end

assign rf_waddr_wb_o = rf_waddr_wb_q;
assign rf_wdata_wb_mux[0] = rf_wdata_wb_q;
assign rf_wdata_wb_mux_we[0] = rf_we_wb_q & wb_valid_q;
```

I feel the aligned version is preferable, though there may be some differences of opinion, perhaps such alignment could be a configurable option?

GregAC, 2020-07-08:
A related issue is the alignment of the LHS of assignment expressions, as an example from https://github.com/lowRISC/ibex/blob/master/rtl/ibex_controller.sv

Before formatting:

```systemverilog
// Decoder doesn't take instr_valid into account, factor it in here.
assign ecall_insn      = ecall_insn_i      & instr_valid_i;
assign mret_insn       = mret_insn_i       & instr_valid_i;
assign dret_insn       = dret_insn_i       & instr_valid_i;
assign wfi_insn        = wfi_insn_i        & instr_valid_i;
assign ebrk_insn       = ebrk_insn_i       & instr_valid_i;
assign csr_pipe_flush  = csr_pipe_flush_i  & instr_valid_i;
assign instr_fetch_err = instr_fetch_err_i & instr_valid_i;
```

After formatting:

```systemverilog
// Decoder doesn't take instr_valid into account, factor it in here.
assign ecall_insn = ecall_insn_i & instr_valid_i;
assign mret_insn = mret_insn_i & instr_valid_i;
assign dret_insn = dret_insn_i & instr_valid_i;
assign wfi_insn = wfi_insn_i & instr_valid_i;
assign ebrk_insn = ebrk_insn_i & instr_valid_i;
assign csr_pipe_flush = csr_pipe_flush_i & instr_valid_i;
assign instr_fetch_err = instr_fetch_err_i & instr_valid_i;
```

Here I'd like to keep the alignment of both the `=` and `&`. This case is trickier than the first case as you may not always want to do it. In this instance I chose to align the `&` up to make it very clear that everything is getting ANDed with the same `instr_valid_i` signal. Were each line to have entirely separate operands I'd still align the `=` but probably not align the operators.

Perhaps one possibility is for the formatter to see that the input has alignment on the '&' and to maintain it on the output? You can of course just turn off formatting with appropriate comments for such blocks but it is preferable to minimise or totally eliminate cases where you do that in the RTL.

fangism, 2020-07-21:
Implementation tip:
For a first cut, I would advise focusing on just aligning the left-hand-side and the equality operator.
It will take a little logic to detect the ranges of consecutive assignment statements.
Aligning on expression operators like `&` is possible, requires more intelligent shape-detection logic, but can be follow-up.

fangism, 2020-07-21:
b/161814377

wtatarski, 2020-08-12:
@fangism 
Is using `TabularAlignTokenPartitions` function a good way for the first part of the issue? I think about applying it to `NodeEnum::kBlockItemStatementList` processing. Something similar is used for `NodeEnum::kPortDeclarationList`

fangism, 2020-08-12:
Yes, `TabularAlignTokenPartitions` on a StatementList is a good starting point.  One difference here is that not all statements are assignments, so you'll need to detect subranges of partitions to apply.

fangism, 2020-08-20:
https://github.com/google/verible/commit/bee424ae46ca2510ef728547d76318b0feca59b6 may make it easier to select subranges of statements that are consecutive assignments.

fangism, 2020-08-29:
Checking in, do you need any help?

commit 0016f3df (2020-09-22): Align blocking and nonblocking assignment statements.

These are controlled by the same policy that applies to continuous assignment
statements.

The existing alignment cell scanner function was extended to work on
the new assignment statement types, not just continuous assignments.

fixes #357

PiperOrigin-RevId: 332959917
closed 2020-09-22 by hzeller