chipsalliance/verible #358: Formatter: Align variable declarations

GregAC, 2020-07-08:
Could the formatter align variable declarations such that the type and dimensions are aligned?

As an example from https://github.com/lowRISC/ibex/blob/master/rtl/ibex_core.sv

Before formatting:
```systemverilog
logic [15:0] instr_rdata_c_id;               // Compressed instruction sampled inside IF stage
logic        instr_is_compressed_id;
logic        instr_fetch_err;                // Bus error on instr fetch
logic        instr_fetch_err_plus2;          // Instruction error is misaligned
logic        illegal_c_insn_id;              // Illegal compressed instruction sent to ID stage
logic [31:0] pc_if;                          // Program counter in IF stage
logic [31:0] pc_id;                          // Program counter in ID stage
logic [31:0] pc_wb;                          // Program counter in WB stage
logic [33:0] imd_val_d_ex[2];                // Intermediate register for multicycle Ops
logic [33:0] imd_val_q_ex[2];                // Intermediate register for multicycle Ops
logic [1:0]  imd_val_we_ex;
```

After formatting:
```systemverilog
logic [15:0] instr_rdata_c_id;  // Compressed instruction sampled inside IF stage
logic instr_is_compressed_id;
logic instr_fetch_err;  // Bus error on instr fetch
logic instr_fetch_err_plus2;  // Instruction error is misaligned
logic illegal_c_insn_id;  // Illegal compressed instruction sent to ID stage
logic [31:0] pc_if;  // Program counter in IF stage
logic [31:0] pc_id;  // Program counter in ID stage
logic [31:0] pc_wb;  // Program counter in WB stage
logic [33:0] imd_val_d_ex[2];  // Intermediate register for multicycle Ops
logic [33:0] imd_val_q_ex[2];  // Intermediate register for multicycle Ops
logic [1:0] imd_val_we_ex;
```

It would be preferable to keep the alignment though the excess whitespace before the comments could go, i.e.

```systemverilog
logic [15:0] instr_rdata_c_id;       // Compressed instruction sampled inside IF stage
logic        instr_is_compressed_id;
logic        instr_fetch_err;        // Bus error on instr fetch
logic        instr_fetch_err_plus2;  // Instruction error is misaligned
logic        illegal_c_insn_id;      // Illegal compressed instruction sent to ID stage
logic [31:0] pc_if;                  // Program counter in IF stage
logic [31:0] pc_id;                  // Program counter in ID stage
logic [31:0] pc_wb;                  // Program counter in WB stage
logic [33:0] imd_val_d_ex[2];        // Intermediate register for multicycle Ops
logic [33:0] imd_val_q_ex[2];        // Intermediate register for multicycle Ops
logic [1:0]  imd_val_we_ex;
```

Would be the preferred output

fangism, 2020-07-21:
Some of the logic in the alignment handler for module port declarations can be used for this task.
A little logic required to identify the ranges of consecutive variable declarations.
Initially, comments could remain unaligned, I'll leave that up to the implementer.
Initial focus is on establishing columns for the dimensions.

fangism, 2020-07-21:
b/161814194

commit fe32ee10 (2020-08-18): Align groups of data/variable/net declarations inside modules.

Grouping is determined by consecutive data/net declarations
regardless of spacing between declarations.

Columns are sliced similarly to port declarations, respecting packed/unpacked
dimensions.

This logic could be re-used inside other contexts as well.

Does not handle inside `ifdef blocks or conditional/loop generate blocks yet.

Does not align "= expr" initializers yet.

Fixes #358

PiperOrigin-RevId: 327281779
closed 2020-08-18 by hzeller