chipsalliance/verible #359: Formatter: Ensure condition is on the same line as ? in ternaries

GregAC, 2020-07-08:
Where the formatting is adding new lines to a ternary expression it should ensure the `?` part of the expression is kept on the same line as the condition. As an example from: https://github.com/lowRISC/ibex/blob/master/rtl/ibex_icache.sv

Before formatting:
```systemverilog
assign prefetch_addr_d     =
    lookup_grant_ic0 ? (lookup_addr_aligned + {{ADDR_W-LINE_W-1{1'b0}},1'b1,{LINE_W{1'b0}}}) :
                       addr_i;
```

After formatting:
```systemverilog
  assign prefetch_addr_d = lookup_grant_ic0
      ? (lookup_addr_aligned + {{ADDR_W - LINE_W - 1{1'b0}}, 1'b1, {LINE_W{1'b0}}}) : addr_i;
```

fangism, 2020-07-08:
Agreed.  This is a matter of penalty tuning.

rkapuscik, 2021-07-20:
I would like to ask for clarification regarding the preferred style. As noted [in this comment](https://github.com/chipsalliance/verible/pull/860#issuecomment-876237068), I read the issue as "keep the question mark part of the expression with the condition" (and #860 fixes this), but this could also be read as "keep the first expression together with the question mark".
The lowRISC style guide is not clear on this matter. 

Proposed behavior is to force the `?` to stay with the condition, then attach the remaining expressions if they fit on the line. Should the first expression after the `?` be instead forced to be on the same line? (if so, [this style guide example](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#expressions) would need an update).

I'm mostly asking for examples other than the [nested ternaries issue](https://github.com/chipsalliance/verible/issues/360) where the desired formatting seems clear - have each successive ternary expression start its own line.

GregAC, 2021-07-29:
> Proposed behavior is to force the ? to stay with the condition, then attach the remaining expressions if they fit on the line.

This sounds sensible to me.

PR #860 (2021-08-18): Keep '?' with condition in ternaries
Fixes #359.
Also fixes #431 - only adds test cases with comments, it seems the indentation issue was resolved some time ago.
closed 2021-08-18 by tgorochowik