chipsalliance/verible #360: Formatter: Align nested ternaries

GregAC, 2020-07-08:
A common construct in verilog is using a nested ternary to express a priority selection structure, the output produced by the formatter for these isn't hugely readable, as an example from: https://github.com/lowRISC/ibex/blob/master/rtl/ibex_icache.sv

Before formatting:
```systemverilog
assign tag_index_ic0 = inval_prog_q   ? inval_index_q :
                       ecc_write_req  ? ecc_write_index :
                       fill_grant_ic0 ? fill_index_ic0 :
                                        lookup_index_ic0;
```

After formatting:
```systemverilog
assign tag_index_ic0 = inval_prog_q ? inval_index_q :
    ecc_write_req ? ecc_write_index : fill_grant_ic0 ? fill_index_ic0 : lookup_index_ic0;
```

In the before version it's very clear what output is associated with what condition, the formatted version is harder to read.

fangism, 2020-07-09:
This makes for an interesting case of shape-detection: looking for a syntax tree shape that is actually linearly nested but formatted in a flat strategy.  It turns out this is what happens already with cascaded if-else-if constructs in 

https://github.com/google/verible/blob/19dade32b36684002ea4f88a9b8786d5d3402b92/verilog/formatting/tree_unwrapper.cc#L588

and

https://github.com/google/verible/blob/19dade32b36684002ea4f88a9b8786d5d3402b92/verilog/formatting/tree_unwrapper.cc#L1176

The handling for this reshapes partitions from the bottom up (last else-if first) and incrementally in the latter function.  

To solve this problem, I would start by tackling token partition tree shaping, and then alignment of those partitions next, using the existing alignment framework.

Since were dealing with expressions that can appear nearly anywhere, in the long run, we should look into bottom-up (dynamic programming) strategies to make this generally applicable in many places.

fangism, 2020-08-11:
@msfschaffner FYI, you mentioned this one

fangism, 2020-08-11:
Also worth noting is that this is potentially a nested alignment problem.  `assign` statements themselves may desire a particular alignment around `=`.

tgorochowik, 2021-10-15:
Continued on #953, closing this ticket

closed 2021-10-15 by tgorochowik