chipsalliance/verible #361: Formatter: Avoid whitespace in bit selection

GregAC, 2020-07-08:
The formatter introduces whitespace around operators inside bit-selections. I'd argue these are better off as compact expressions, I note whitespace isn't added in declarations.

For example in https://github.com/lowRISC/ibex/blob/master/rtl/ibex_icache.sv

Before formatting:
```systemverilog
logic [INDEX_W-1:0]                  inval_index_d, inval_index_q;
```

After formatting:
```systemverilog
logic [INDEX_W-1:0] inval_index_d, inval_index_q;
```

The extra whitespace between the dimension and the names is removed but the expression inside the `[]` remains compact.

Before formatting:
```systemverilog
assign lookup_addr_aligned = {lookup_addr_ic0[ADDR_W-1:LINE_W],{LINE_W{1'b0}}};
```

After formatting:
```systemverilog
assign lookup_addr_aligned = {lookup_addr_ic0[ADDR_W - 1:LINE_W], {LINE_W{1'b0}}};
```

Similar syntax (though used to select bits rather than declare dimensions) but this time whitespace is added around the `-`

fangism, 2020-07-09:
Confirming the behavior you observe.

Reference:
https://github.com/google/verible/blob/19dade32b36684002ea4f88a9b8786d5d3402b92/verilog/formatting/token_annotator.cc#L389

https://github.com/google/verible/blob/19dade32b36684002ea4f88a9b8786d5d3402b92/verilog/formatting/token_annotator.cc#L294

What is happening here is that we've decided for the time being to limit spacing around `:` to 1 space maximum _in declarative range contexts_ like ports, and symmetrize the right spacing based on the original spacing to the left of the `:`.  The style guide we've been implementing so far is a parent of the lowRISC guide, and we only decided to allow compaction in some context, but decided not to force 0-spacing in the formatter tool at this time -- in other words, we've partially punted with the current half-solution.  

https://github.com/google/verible/blob/19dade32b36684002ea4f88a9b8786d5d3402b92/verilog/formatting/token_annotator.cc#L689

The `Preserve` here means, "keep the original spacing", which explains why some spaces are untouched.

Indexing/bit-slicing have not been singled out as special cases at this time, so by default the spacing around binary operators is 1 (see near the above code excerpts).  But this would be easy to do because the formatter knows the full syntactic context of every token.

ldalek-antmicro, 2020-10-30:
Hi @fangism 
I've created PR that forces compaction of expressions in bit selection ranges: #558 
I'm not sure what to do with ranges in declaration contexts (leaved as it is right now).

Also I've noticed that master branch is correctly preserving spaces in example expressions - only limiting to 1 space and symmetrize ranges:
<pre>
assign lookup_addr_aligned = {lookup_addr_ic0[ADDR_W-1:LINE_W],{LINE_W{1'b0}}};
</pre>
formats like this:
<pre>
assign lookup_addr_aligned = {lookup_addr_ic0[ADDR_W-1:LINE_W], {LINE_W{1'b0}}};
</pre>

fangism, 2020-10-31:
You can leave declarative dimensions as-is.

I re-tested top-of-tree with:

```systemverilog
module m;
assign lookup_addr_aligned = {lookup_addr_ic0[  ADDR_W  -  1  :  LINE_W  ],{LINE_W{1'b0}}};
endmodule : m
```

and got

```systemverilog
module m;
  assign lookup_addr_aligned = {lookup_addr_ic0[ADDR_W - 1 : LINE_W], {LINE_W{1'b0}}};
endmodule : m
```

I think your change will result in:

```systemverilog
module m;
  assign lookup_addr_aligned = {lookup_addr_ic0[ADDR_W-1 : LINE_W], {LINE_W{1'b0}}};
endmodule : m
```

Can you confirm?

If so, that looks ok to me.  
In the long term, this is potentially something users may want to configure, but we haven't worked out a configurability story yet.

I can imagine users may have different opinions on spacing in contexts like this.

I'll comment in the PR.

ldalek-antmicro, 2020-11-03:
Not exactly. My changes 'forces' compaction of the whole dimension range expression. So for input like:
```systemverilog
module m;
assign lookup_addr_aligned = {lookup_addr_ic0[  ADDR_W  -  1  :  LINE_W  ],{LINE_W{1'b0}}};
endmodule : m
```
I got:
```systemverilog
module m;
  assign lookup_addr_aligned = {lookup_addr_ic0[ADDR_W-1:LINE_W], {LINE_W{1'b0}}};
endmodule
```

I assumed that intentention of this issue is to force compaction of the whole expression and not just a binary operators.

fangism, 2020-11-03:
For now, I suggest limiting it to binary operators.  Unary operators are already compact.

What else is left?  call like expressions like `$builtin(a, b)`.  

It's arguable whether `:` is a binary operator because it means "range".  I suggest leaving spacing around `:` as-is (unaffected) for now.

commit 4fc56cdc (2020-11-04): PR #558: Compact bit selection/indexing expressions
Fixes #361

GitHub PR https://github.com/google/verible/pull/558

Copybara import of the project:

  - dc11e00f8229b740783387305acebaa8c0019bf9 Compact expression inside bit slice selection/indexing op... by Lukasz Dalek <ldalek@antmicro.com>
  - 266d8d9a1df7d7d23e43992ec12804ef757d4f10 Merge dc11e00f8229b740783387305acebaa8c0019bf9 into fd031... by Lukasz Dalek <48219411+ldalek-antmicro@users.noreply.github.com>

Closes #558

PiperOrigin-RevId: 340539919
closed 2020-11-04 by fangism