chipsalliance/verible #365: Use case: Removing / expanding preprocessor statements and/or parameters

WeeBull, 2020-07-15:
I'm sorry if this is the wrong forum, but I'm trying to understand what verible is capable of.

I have a need to process SV files and remove code which is unused in a certain configuration - configurable IP being delivered to a customer with a specific configuration applied, and then dead code stripped.

I note that the input to verible is "_un-preprocessed source files_". Would verible be able parse a set of source files with either defines or  parameters set on the command line, and output new files with those values propagated through the code? Defines seem like the simpler case, but given a full AST is being generated a parameter might not be that hard either.

Has anybody already addressed a use case like that?
Would verible be the right basis on which to build functionality like that if not?

mithro, 2020-07-15:
You might want to check if https://github.com/alainmarcel/Surelog is a better fit for your project.

fangism, 2020-07-15:
It sounds like you want the equivalent of `cpp` preprocessing via a standalone SV preprocessor, mainly to strip away conditional text.  (A common convention among tools is to use `-E` to save the preprocessed output, before piping that result to the underlying compiler.)

I would first look into Verilator's preprocessor for the job, I think they may have a high-quality preprocessor.

I had plans for a pseudo-preprocessor, but have not gotten around to making significant progress on it.
https://github.com/google/verible/tree/master/verilog/preprocessor

fangism, 2020-07-15:
I've been known to hijack `cpp` for Verilog purposes: the trick is to turn Verilog preprocessing directives into ones that `cpp` understands, and be able to reverse the transformation back into Verilog.  So verilog->C->cpp->C->verilog.

I use this script to strip comments from Verilog files:

```bash
#!/bin/bash
awk '
BEGIN {continued=0;}
/\/\// { if(continued) sub("//", "__PRESERVE_COMMENT__ ");}
/\\$/ {continued=1; print;}
!/\\$/ {continued=0; print;}
' | \
sed -e "s|'| __SINGLE_QUOTE__ |g" \
    -e 's|\\$|__LINE_CONTINUATION__|g' \
    -e 's|\#\#|__DOUBLE_POUND__|g' \
    -e 's|\#|__SINGLE_POUND__|g' \
    -e 's|`__FILE__|__VERILOG_FILE__|g' \
    -e 's|`__LINE__|__VERILOG_LINE__|g' | \
cpp -P | \
sed -e 's|__VERILOG_FILE__|`__FILE__|g' \
    -e 's|__VERILOG_LINE__|`__LINE__|g' \
    -e 's|__SINGLE_POUND__|#|g' \
    -e 's|__DOUBLE_POUND__|##|g' \
    -e 's|__LINE_CONTINUATION__$|\\|g' \
    -e "s| __SINGLE_QUOTE__ |'|g" \
    -e 's|__PRESERVE_COMMENT__ |//|'
```

(This script is not perfect, but gets pretty far.)

If you manipulate `` `ifdef `` to look like `#ifdef` and forward `-D` options to behave like `#define` you might be able to come up with something.

WeeBull, 2020-07-17:
Thanks for the feedback.

I wasn't wanting to go for `-E` type preprocessing because that expands everything, making code largely unreadable. I'm looking for something where I could expand things related to 1 or 2 defines and leave everything else intact.

I'll give Surelog a look.

fangism, 2020-07-17:
Partial expansion and evaluation is exactly what our pseudo preprocessor was intended to do.

I'm going to keep this one open, even if you find an alternate solution.  This would make for a clean standalone project that already has a head start.

The abstraction of the problem here is a transformation of one lexical token stream into another.
The scanner (current pseudo preprocessor) could be a state machine with a configurable initial conditions and action pathways that could filter/replace text.  If we're only tackling preprocessor conditional evaluation, then filtering is enough.

WeeBull, 2020-07-20:
That's fine. I must admit I've gone to look at your log and and the fact is a preprocessor that will expand everything means it's not suitable for my use case. I think the fact that variable will create a syntax tree that contains the preprocessor commands means it's a lot closer to what I need.

fangism, 2020-07-20:
Just clarifying:
The work-in-progress pseudo-preprocessor doesn't touch any syntax tree, it is tokens-in, tokens-out (which may then feed a parser).  This approach is guaranteed to work for any lexically correct input file (need not be syntactically clean).

The syntax tree produced by the parser does support "sane" placement of preprocessing directives -- e.g. conditionals around sequences of whole items.  A typical example that does NOT work with this approach is:

```systemverilog
`ifdef NAME_IT_FOO
module foo;
`else
module bar;
`endif
endmodule
```

fangism, 2020-07-27:
Related discussion about partial evaluation going on in #366