chipsalliance/verible #366: macro expansion

thingkingland, 2020-07-15:
For a legal Verilog file:

    `define A t;
    module test`A
    wire a;
    wire b;
    assign a = b;
    endmodule

Verible parser rejects it with

test1.v: test1.v:2:12: syntax error, rejected "`A" (https://github.com/google/verible).
test1.v:5:1: syntax error, rejected "assign" (https://github.com/google/verible).
test1.v:6:1: syntax error, rejected "endmodule" (https://github.com/google/verible).

while tools like Icarus Verilog accepts it. I take a look at the C++ code, and it seems to me that it parses without macro expansion first then calls function "VerilogAnalyzer::ExpandMacroCallArgExpressions()" to expand each parsed macro tokens. This assumes macros without expansion must satisfy SystemVerilog syntax. Should macros be expanded before syntax is checked?

fangism, 2020-07-15:
Verible doesn't include a proper preprocessor at this time, and is thus limited in where it can allow macros (grammatically).  Depending on your intent, you could preprocess first before feeding to Verible's parser/tools.

thingkingland, 2020-07-27:
Hi David,

Thanks for the reply. I make an attempt to enhance Verible code to support `define, `undef, `undefineall without a separate preprocessor step. The solution can be extended to support `include and `ifdef. The solution will remove these directives from tokens. This should be good for normal EDA applications such as simulation and synthesis. For formatter, those directives may need to be preserved. My current implementation uses a switch to choose whether to expand those directives or to keep them in the current way. There should be a way to encode both in one token list, if the formatter can utilize both information.

I have been developing the solution in my private git branch. If you think there might be value to incorporate this solution into public Verible code, please let me know and I may volunteer as a contributor to Verible.

fangism, 2020-07-27:
I'm excited you started this effort.  Did you happen to take a look at 
https://github.com/google/verible/blob/master/verilog/preprocessor/verilog_preprocess.h ?
That already contained some code for tracking `` `define``s, but doesn't attempt to apply/expand them yet.

What I've been thinking is assembling a suite of pseudo-preprocessor like functions.  This first one could be called something like `expand-known-macro-definitions`.  This would be available both as a library function and standalone tool (`verible-verilog-pseudo-preprocess` ??).  

Ultimately, I would like for preprocessing transformations to maintain a full trail of 'interpretations' to be able to trace interpreted text to its original locations.  This 'integrated' preprocessing is more powerful than string-in-string-out +FILE/LINE directives, and leads to better diagnostics.  I sort of envision this like expandable tree view of tokens.  And of course, formatting would only ever operate on operate on original text and tokens, but analyses like style linting could have the option parsing various interpretations.  

How about we continue to spec out the proposed changes including test cases?

Strawman proposal:

* inputs: raw token stream with preprocessing directives, plus possibly a set of initial defines (like `+define`) supplied by the user.
* output: a possibly transformed tokenstream with some preprocessing directives removed, and possible some macros substituted with their interpreted equivalent tokens (concatenations would replace multiple tokens).  This depends on the selection functionality, but we can start with `expand-known-macro-definitions`.  The resulting token stream could then be assembled back into text/string.
* unknown macro definitions at the time of evaluation will be left intact
* `` `include``s will not be expanded
* `` `define``s will be evaluated and tracked and removed
* `` `ifdef `` and such will be evaluated and removed
* nested conditionals/defines inside conditionals/defines will (???) be evaluated

thingkingland, 2020-07-27:
Hi David,

If it is ok for you to have private conversation on this topic, please send an email to rd@thingkingland.app <mailto:rd@thingkingland.app>
I can email you my sample code and discuss more. If you want to continue on GitHub public discussion, that is fine with me too.

I read the existing code which attempts to expand a macro token after the lexer. My solution is to do it inside lexer with yypush_state_buffer. The reasons include

1. As you wrote in verilog_preprocess.h:
// TODO(fangism): token concatenation, e.g. a``b
   plus the removal of backslash before \n, and the removal of one-line comments (//) in macro text,
   macro expansion creates tokens whose text is not a contiguous region of the original text, and it cannot be represented by a string_view for each token.

2. Macro expansion needs to be recursive if the result contains another `MacroId. If we use “AnalyzeVerilogConstruct” in existing code to create a new lexer for each `macro, it may introduce recursive calls of “AnalyzeVerilogConstruct” and reentrant issues of lexer need to be dealt with. My solution uses yypush_state_buffer for recursion inside lexer, not after.

3. Though I have not implemented `include, `include should be handled together with `define. It is very likely that `include file may contain `define, and it does not make sense to separate them. yypush_state_buffer can handle `include too.

I only looked into verible code for only 3 weeks. Please forgive me if I misunderstand too much.

The intent of verible parser to preserve information of the original text is different from most other SystemVerilog parsers I have seen, but I think it is a good feature and it enables capabilities other parsers cannot do. I have an idea to combine both. Suppose a macro usage

               `FOO

is expanded to a list of tokens:

               T0, T1, …., Tn

If we prepend and append two special dummy tokens:

              <beginFoo>T0, T1, …., Tn<endFoo>

and let "VerilogLexer::KeepSyntaxTreeTokens” function ignore <beginFoo> and <endFoo> for the parser, the parser sees tokens without macros, while the formatter should be able to recover `FOO by skipping tokens between <beginFoo> and <endFoo>. The dummy tokens only need to be inserted at top level macros, and recursive macro expansion does not need them. The formatter only need information on original top-level macros.

fangism, 2020-07-27:
Sent you a message, but I can answer a few topics here that others may be interested in.

1. For a "normal" production compiler-like toolchain, doing this work in the lexer might be appropriate, but we must consider that the lexer needs to be suited for applications that operate on un-preprocessed text, like the formatter, and even style-linter (which may want to analyze matters pertaining to preprocessor usage).  Thus, I would like to keep the lexer as singularly-functioned (just produce tokens) as much as possible and push additional functionality into separate passes.  One example of this is https://github.com/google/verible/blob/master/verilog/parser/verilog_lexical_context.h, which is a pass between lexing and parsing that disambiguates tokens with multiple meanings like `->`, and only mutates a token stream.  This is the main reason why I want to treat all preprocessing-like operations as a function of token streams.

2.  Absolutely.  The dynamic nature of proper Verilog preprocessing necessitates that expansion also includes evaluation -- they cannot be decoupled, _unless_ you are intentionally doing partial/approximate/selective evaluation, which falls into the category of "features other tools typically do not provide".  However, I think this can be abstracted as operations on a token stream (post-lexer).  An example of an operation that operates on the lexer output (token stream) is equivalence checking (to make sure the formatter didn't butcher the text), which is done in https://github.com/google/verible/blob/master/verilog/analysis/verilog_equivalence.cc.  In today's lexer, macro definition bodies are one unlexer atom of text which is suitable for parsing, but the comparison actually recursively lexes the definition body to inspect its contents as well.  (The formatter does not reflow things like comments and macro definition bodies... yet.)

3.  I consider macro and include expansion part of the same class of creating _interpreted_ views of original text.  And different applications may want to parse different lexical interpretations.  Dummy-tokens is one possible manifestation of selectively interpreting text.  I would extend your example to also include macros with arguments, like `` `FOO(x, y, z) ``.  Instead of expanding one token, you'd actually be providing alternate interpreted views over a _sub-range_ of tokens (since macro calls are tokenized down to their positional arguments today).  Either way, a single or multiple tokens can be a unit with multiple interpretations (expansions).

Today's tokens and token stream are constrained in that the string_views of the individual tokens must be a sub-string (range) of the original full text body.  This lets us efficiently calculate and report byte offsets.  A lot of other operations are built around this invariant.  Operations like token-concatenation would produce new strings which would have to be stored somewhere to maintain correct object lifetime.  The formatter (parsing) is expected to never expand/concatenate tokens, but a style-linter might.  For streaming operations that don't need long-lived temporary tokens (like printing), this may still be ok, but for persistent transformed token streams (including trees/views), more work may be needed.