chipsalliance/verible #368: Wrong indentation on module instantiations spanning multiple lines

zegervdv, 2020-07-16:
When instantiating a module with many ports it is often split over multiple lines.
The first line of the instance is indented correctly but the continuation is indented one less:
```systemverilog
module foobar (
    input logic foo
);

  inner_foo #(
    .p_option (4)
) foo (
    .foo (foo),
    .bar (bar)
);

endmodule
```

Expected:
```systemverilog
module foobar (
    input logic foo
);

  inner_foo #(
      .p_option (4)
  ) foo (
      .foo (foo),
      .bar (bar)
  );

endmodule
```

It seems like it is not touching the continued lines at all.
For example for this input:
```systemverilog
module foobar (
    input logic foo
);

inner_foo #(
.p_option (4)
) foo (
.foo (foo),
        .bar (bar)
);

endmodule

```
The result  is:
```systemverilog
module foobar (
    input logic foo
);

  inner_foo #(
.p_option (4)
) foo (
.foo (foo),
        .bar (bar)
);

endmodule

```

fangism, 2020-07-16:
Ah, I realized that for the time being we've disabled formatting around module instantiations in the short term because users were complaining about lack of alignment or ruining pre-existing alignment.  This is controlled via

`verible-verilog-format --format_module_instantiations`

(The default is false [off], which preserves existing spacing in instantiations.)
If I re-run your examples with that flag enabled, you'll get:

```systemverilog
module foobar (
    input logic foo
);

  inner_foo #(
      .p_option(4)
  ) foo (
      .foo(foo),
      .bar(bar)
  );

endmodule

module foobar (
    input logic foo
);

  inner_foo #(
      .p_option(4)
  ) foo (
      .foo(foo),
      .bar(bar)
  );

endmodule
```

Apologies for the temporary inconvenience.  We are working alignment matters.

zegervdv, 2020-07-17:
I missed that setting. It seems to work fine for me now.

fangism, 2020-07-17:
Thanks for confirming.

closed 2020-07-17 by fangism