chipsalliance/verible #369: Interfaces in port mapping misaligned

zegervdv, 2020-07-16:
Adding an interface in the port mapping of a module messes up the formatting:

```systemverilog
module foobar #(
    int foo
) (
    input logic          bar,
          itf.master baz
);

endmodule
```

Also note the space that is added between the `) (`

fangism, 2020-07-16:
b/160634864

fangism, 2020-07-16:
It turns out I actually just fixed this a few days ago, and there is a commit in the queue waiting to be pushed.  @hzeller can you push?

fangism, 2020-07-16:
As for `) (`, the formatter is following guidance that may be slightly different from lowRISC's.  Please file that issue separately.  I may not have the bandwidth to introduce configurability, but I could point you to the code that affects this:

https://github.com/google/verible/blob/1886f72085356d3169d35a8b496f97fe1c662959/verilog/formatting/token_annotator.cc#L482

zegervdv, 2020-07-16:
I was under the impression the `) (` only happened when I had an interface in the port map, but that is not the case.
I don't really mind about the space as long as it's consistent :)

fangism, 2020-07-16:
> I was under the impression the `) (` only happened when I had an interface in the port map, but that is not the case.
> I don't really mind about the space as long as it's consistent :)

The fewer {rules/priorities/context-sensitivities/special cases} have to be coded up in the function I linked, the more consistent things will be.  :)

fangism, 2020-07-20:
Fixed by https://github.com/google/verible/commit/ebdb13d763664eea1a462d6f635d8872a7051190

closed 2020-07-20 by fangism