chipsalliance/verible #370: check declared identifiers against set of unwanted patterns

fangism, 2020-07-22:
Task: implement a style lint rule that can flag a _declared_ identifier if it matches a set of configurable patterns

The key here is that the syntax tree tells you exactly which identifiers are declared vs. referential.

Examples:

* `module m;` : `m` is declared
* `mymodule myinst(...);` : `myinst` is declared, `mymodule` is referential
* `function f;` : `f` is declared
* `f(...)` : `f` is referential
* `foo_pkg::mytype_t var_name` : only `var_name` is declared, others are referential

Suggestions: 

* The number of declaration contexts if finite and reasonable, so explictly check for those, and assume all others (unhandled) are referential.
* Use a SyntaxTreeLintRule to find every declaration that introduces a new symbol.

fangism, 2020-07-22:
b/161864321

muhdmud, 2020-07-28:
good day all, can i work on this?

muhdmud, 2020-07-29:
thanks for the opportunity. this is my first time and i just applied for the CLA.

while waiting if you could point me in the right direction, looks like i have to look at verilog/analysis/checkers folder to see how rules are being implemented?

correct me if im wrong

fangism, 2020-07-29:
Yes, verilog/analysis/checkers/*name_style_rule.* is a good place to start because those are rules about identifiers.

While many of those rules are implemented using Matchers, you don't necessarily have to use them, in fact, I would advise against them.

I think you'll end up needing to write CST accessor functions like:
https://github.com/google/verible/blob/74b906bcefefbda3dea36abba4a6a2ce5707d5f0/verilog/CST/module.h#L67

for a variety of declarative constructs.

Most of the rule work will probably be done in the `MyNewRule::Visit(const SyntaxTreeNode&)` function that you write.

Roughly pseudo-coded something like:

* get node's tag, which tells you which syntax construct you're looking at
* switch on that tag
* cases will cover all constructs that declare something (for now, don't worry about implicit declarations of things like signals)
* for each case, call a CST function that extracts the declared name
* check that declared name against set of unwanted patterns, and report a violation at the offending sites

Does this help?

fangism, 2020-07-29:
Another suggestion is to come up with test cases first.

For example:
https://github.com/google/verible/blob/master/verilog/analysis/checkers/constraint_name_style_rule_test.cc

In particular, `LintTestCase` let you mark interesting pieces of text where you expect to find violations (non-overlapping ranges).

muhdmud, 2020-07-29:
ok give me a couple days to have a look and will ask here if any doubts. Thanks so much for the pointers!

fangism, 2020-07-29:
Also, don't be afraid to send PRs for work-in-progress.  If you write some new CST helper functions with tests (and you may have to search around in CST/ to discover which already exist), those can be reused by others without waiting for the whole task to be completed.

muhdmud, 2020-07-30:
Alright noted! i am looking at the systemverilog syntax for better understanding. 
for the cst accessor functions, an example i can see that for identifier_test.cc tests, in this case those without scope operator are being flagged as unqualified.

in my case, i need to create cst accessor something like 
const (return some construct here?) & GetDeclarativeConstruct(const verible::Symbol&)
this will recurse the syntax tree and get the constructs?

then tests will be a list of lint testcases like the one you showed above,
"module m; " : m is declared
"function f; " : f is declared

TEST(checkDeclaredIdentifier, simpletest) will have above strings and we pass this into verilog analyzer.

Correct me if im wrong, trying to understand flow of cst helpers and their tests!
regarding the PR, i can just write 3 WIP files, like declartiveconstruct.cc, .h and _test.cc?

fangism, 2020-07-30:
Yes, some declarations can bear both unqualified and qualified identifiers.  For example, out-of-line method definitions like:

```systemverilog
function void foo_class::bar_function (...);
```

So the returned subtree could be either type, not necessarily a leaf node.  Since out-of-line definitions can only follow corresponding in-line declarations, it's not as important to verify the last component of the qualified name (`B` in `A::B`) case.  You would just see the same name being flagged in two places (which is ok too).

Instead of an all-inclusive `GetDeclarativeConstruct()` that covers all declarations, I would focus on one element type at a time.  

Possible warm-up exercise (just a suggestion):

Convert the tests in verilog/CST/package_test.cc into the style used in verilog/CST/declaration_test.cc where test cases are marked up with regions that correspond to expected subtrees of interest.  This style of testing is more precise than just checking that the name matches a string-literal; it actually verifies the position within the original source code of the occurrence of text.  (CST/declaration.h gives some good examples of writing some helper functions to FindAllSomeTreeNodeType which are handy for testing.)

muhdmud, 2020-07-30:
Yes that would be great, i will work on this warmup exercise. 

looks like for package_test.cc, one string literal is being passed into the analyzer, one for each test, while for declaration_test.cc, a whole bunch of kTestCases are being passed into the test. 
Also instead of having exact matches like in package.cc, the declaration.cc have helper functions to get subtree(so there is more information?) instead of exact tree match.

Also means i need to come up with multiple testcases for package declarations, and first write some helper functions in package.cc that fetches these subtrees.

Am i correct?

fangism, 2020-07-30:
Correct, package_test.cc (and many others like it) have separate hand-coded checks for each `TEST()` test case (string literal input), which is cumbersome to maintain, compared to using the `SyntaxTreeSearchTestCase` construct, which supports checking against any N expected regions of interest, handy for test cases with multiple declarations.

package.cc already has a `FindAllPackageDeclarations` function, which you can then use with `GetPackageNameToken` to construct an array of found names, and compare against expected ones.
This would strengthen:

https://github.com/google/verible/blob/617b40419adc1c39f5ad553c0ec50843abd0fb1a/verilog/CST/package_test.cc#L75

which currently only checks that there are 2 package names, but doesn't actually verify what they are, or their text location.

muhdmud, 2020-07-31:
okay noted sir, il write one test in package_test.cc with some test cases and submit a pr. how do i know that its working? or dont worry and just submit this work-in-progress first.

also do I edit the test TEST(FindAllPackageDeclarationsTest, MultiPackages), or i can make another test?

1. Come out with some test cases in an array
2. loop through this array and run FindAllPackageDeclarations on all tests, storing all match in a vector A
3. loop through vector A and run getpackagenametoken and store in another vector B
4. check this vector B with a expected test file

please correct me if my understanding is wrong anywhere

fangism, 2020-07-31:
`bazel test -c opt //...` will run all tests.  (For fun and education, you could intentionally make any test fail and see how that looks.)

You may change `TEST(FindAllPackageDeclarationsTest, MultiPackages)`.  Doesn't have to be an array the first time, but once you see how similar tests look using the new pattern, you'll find that placing them into an array should be mechanically easy (step 1).

Steps 2-4 sound good for each test case.

muhdmud, 2020-07-31:
okay thanks! il have a look with simpler cases. by the way, is this the preferred mode of communication in open source? still learning on these habits/ways

fangism, 2020-07-31:
This works fine, a lot of this public discussion is useful to other new developers as well.  

https://groups.google.com/g/verible-dev is also good for general discussion.

muhdmud, 2020-07-31:
oh i see, thanks David!

muhdmud, 2020-07-31:
faced below errors when building, are these alright for now?

PS C:\openSource\verible> bazel version
Starting local Bazel server and connecting to it...
Build label: 3.4.1
Build target: bazel-out/x64_windows-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Tue Jul 14 06:32:11 2020 (1594708331)
Build timestamp: 1594708331
Build timestamp as int: 1594708331
PS C:\openSource\verible> bazel build -c opt //...
INFO: Analyzed 1238 targets (75 packages loaded, 5976 targets configured).
INFO: Found 1238 targets...
INFO: From Compiling common/util/logging.cc:
C:\users\muhd\_bazel_muhd\436imxx5\execroot\com_google_verible\common\util\logging.cc(213) : warning C4722: 'verible::internal::LogMessageFatal::~LogMessageFatal': destructor never returns, potential memory leak       
ERROR: C:/opensource/verible/bazel/BUILD:20:15: error executing shell command: 'C:/Users/muhd/scoop/apps/msys2/current/usr/bin/bash.exe -c function cleanup_function() {
local ecode=$?
if [ $ecode -eq 0 ]; then
cleanup_on_success
else
cleanup_on_failure
fi
}
set -e
function cle...' failed (Exit -1). Note: Remote connection/protocol failed with: execution failed
Action failed to execute: java.io.IOException: ERROR: src/main/native/windows/process.cc(202): CreateProcessW("C:\Users\muhd\scoop\apps\msys2\current\usr\bin\bash.exe" -c "function cleanup_function() {
local ecode=$?
if [ $ecode -eq 0 ]; then
cleanup_on_success
else
cleanup_on_failure
fi
}
set -e
function cleanup_on_success() {
printf \"rules_foreign_cc: Cleaning temp directories\"
  rm -rf $BUILD_TMPDIR $EXT_BUILD_DEPS
}
function cleanup_on_failure() {
printf \"\nrules_foreign_cc: Build failed!\nrules_foreign_cc: Keeping temp build directory $BUILD_TMPDIR and dependencies directory $EXT_BUILD_DEPS(...)): The system cannot find the file specified.
 (error: 2)
INFO: Elapsed time: 66.858s, Critical Path: 6.59s
INFO: 35 processes: 35 local.
FAILED: Build did NOT complete successfully

fangism, 2020-07-31:
Oh, I'm actually not familiar with Windows builds.
What compiler are you using?
I wonder if it unable to honor `ABSL_ATTRIBUTE_NORETURN`?  But it seems that is only a compiler warning.

The real problem seems to be coming from bazel build rules.
We'll need to web search for known issues.  :-(

fangism, 2020-07-31:
Suggest looking around:

https://docs.bazel.build/versions/master/windows.html

and file an issue with details referencing your error above:

https://github.com/bazelbuild/bazel/issues?q=is%3Aopen+is%3Aissue+label%3Ateam-Windows (New Issue)

(follow their issue template)

I hope we can unblock you soon...

muhdmud, 2020-07-31:
oh ok thanks David, i will have a look and raise the issue!

muhdmud, 2020-07-31:
#11881 have raised issue with bazel, couldnt find any info in discussion group and in current issues

fangism, 2020-07-31:
Re-linking: https://github.com/bazelbuild/bazel/issues/11881

muhdmud, 2020-08-05:
have tested again to confirm, and still give me some issues.
I did test with a separate sample project, and build worked fine.
Anyway i can circumvent this? or work on other stuff

fangism, 2020-08-05:
Chatting around, some other suggested trying out Linux on Windows

* https://docs.microsoft.com/en-us/windows/wsl/install-win10
* https://ubuntu.com/tutorials/ubuntu-on-windows

Willing to try that?  (Makes a lot of open-source development easier.)

muhdmud, 2020-08-05:
Yeah sure! willing to try anything and learn.
Okay i will have a look and setup using the links you provided, thank you!

muhdmud, 2020-08-07:
managed to set it up using that first link, below is the output after successful build.
are these warnings alright for now? 

muhd@mud:~/opensource/verible$ ls
AUTHORS  BUILD  CONTRIBUTING.md  Docker  LICENSE  README.md  WORKSPACE  bazel  bazel-bin  bazel-out  bazel-testlogs  bazel-verible  common  verilog
muhd@mud:~/opensource/verible$ bazel build -c opt //...
Starting local Bazel server and connecting to it...
INFO: Analyzed 1238 targets (76 packages loaded, 5896 targets configured).
INFO: Found 1238 targets...
INFO: From CcConfigureMakeRule bazel/m4/include:

INFO: From CcConfigureMakeRule bazel/m4/include [for host]:

INFO: From CcConfigureMakeRule bazel/flex/include:

INFO: From CcConfigureMakeRule bazel/flex/include [for host]:

INFO: From Generating verilog/parser/verilog.yy.cc from verilog/parser/verilog.lex [for host]:
verilog/parser/verilog.lex:929: warning, rule cannot be matched
INFO: From Generating common/analysis/command_file.yy.cc from common/analysis/command_file.lex [for host]:
common/analysis/command_file.lex:72: warning, rule cannot be matched
INFO: From Generating verilog/parser/verilog.yy.cc from verilog/parser/verilog.lex:
verilog/parser/verilog.lex:929: warning, rule cannot be matched
INFO: From CcConfigureMakeRule bazel/bison/include:

INFO: From CcConfigureMakeRule bazel/bison/include [for host]:

INFO: From Generating common/analysis/command_file.yy.cc from common/analysis/command_file.lex:
common/analysis/command_file.lex:72: warning, rule cannot be matched
INFO: From Generating verilog/parser/verilog.yy.cc from verilog/parser/verilog.lex:
verilog/parser/verilog.lex:929: warning, rule cannot be matched
INFO: From Generating common/analysis/command_file.yy.cc from common/analysis/command_file.lex:
common/analysis/command_file.lex:72: warning, rule cannot be matched
INFO: From CcConfigureMakeRule bazel/m4/include:

INFO: From CcConfigureMakeRule bazel/flex/include:

INFO: Elapsed time: 622.428s, Critical Path: 216.72s
INFO: 2117 processes: 2117 linux-sandbox.
INFO: Build completed successfully, 2574

fangism, 2020-08-07:
We see those warnings too, you are good to go!

fangism, 2020-08-07:
@MinaToma and @ivan444 FYI, this work is somewhat related to your project in that it is looking for identifiers tokens in specific locations, in this case declarations.  I think there is an opportunity to share `GetElementNameToken`-like functions added to the `CST/` library.

muhdmud, 2020-08-09:
Alright cool, thanks! Il reread what we discussed and start on pr for the testcases

muhdmud, 2020-08-09:
i tried to include "common/analysis/syntax_tree_search_test_utils.h" in package.h since i wanted to have an array of SyntaxTreeSearchTestCase, but it gives me an error below that the no such file, am i missing something here? 

INFO: Analyzed 1238 targets (0 packages loaded, 0 targets configured).
INFO: Found 1238 targets...
ERROR: /home/muhd/opensource/verible/verilog/CST/BUILD:486:11: C++ compilation of rule '//verilog/CST:package' failed (Exit 1) gcc failed: error executing command /usr/bin/gcc -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG -ffunction-sections ... (remaining 26 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
In file included from verilog/CST/package.cc:15:
./verilog/CST/package.h:28:10: fatal error: common/analysis/syntax_tree_search_test_utils.h: No such file or directory
   28 | #include "common/analysis/syntax_tree_search_test_utils.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
INFO: Elapsed time: 0.477s, Critical Path: 0.16s
INFO: 0 processes.
FAILED: Build did NOT complete successfully

also, can i raise a pr so you are able to see the code? or should i fix this first

fangism, 2020-08-09:
When you include a header, you also need to add in the BUILD file a `deps = [...]` to the library from which the header came, in this case, `//common/analysis:syntax_tree_search_test_utils`.

fangism, 2020-08-09:
And yes feel free to open a PR for early review.

Another note: libraries named `test_util` are typically only used for tests of libraries, not the libraries themselves.

muhdmud, 2020-08-10:
okay noted, currently facing some issues when trying to push my branch, due to subsystem permission with ssh,
trying resolve with this https://github.com/microsoft/WSL/issues/3181
and this https://github.com/Microsoft/WSL/issues/81

muhdmud, 2020-09-04:
apologies, got caught up with work. Il review and start WIP pr

muhdmud, 2020-10-09:
Sorry for the delay sir! i'm in midst of job transition, il be back in a month or so!