chipsalliance/verible #372: Use variable before declaration

stephenry, 2020-07-23:
In the example below, `foo` is assigned before it is declared and is therefore infered as the 'default net type' according to 6.10, even if the eventual type is not the default net type. SV2V picks this up and causes the generated result to be incorrect; generates a wire. Yosys subsequently raises a warning as the wire is assigned from the sequential block. Verible should flag instances such as this where the variable use preceeds the declaration.

```systemverilog
always_comb foo = 'b1;
logic foo;
```

fangism, 2020-07-23:
Someone was working on this in #217 
Duplicate of #217

fangism, 2020-07-23:
Welcome to join the conversation there and comment on the #244

closed 2020-07-23 by fangism