chipsalliance/verible #373: Expose an UHDM compatible API?

mithro, 2020-07-24:
As part of the [Surelog project](https://github.com/alainmarcel/Surelog) (a /potentially/ competing ANTLR4 based SystemVerilog parser) we have been trying to create a system which allows different parsers be interfaced to multiple different tools (like Yosys, Verilator, Icarus, etc). 

The attempt at doing this project is [called Universal Hardware Data Model (UHDM)](https://github.com/alainmarcel/UHDM) and structured kind of like this;
[image]

It would be awesome to see more parsers trying to support this API and it would be good for Verible to be such a parser.

See https://github.com/alainmarcel/UHDM/issues/319

mithro, 2020-07-24:
This is a big task that I don't think we have the resources to tackle at the moment, but I wanted to log the issue incase someone did want to start exploring what this might look like.

fangism, 2020-07-24:
Noted.  

There is quite a large gap between what Verible offers today and what is needed to intercept UHDM.
Isn't UHDM a elaborated representation?

Before we can even talk about elaboration, there is semantic analysis.

Before semantic analysis, we at least need a AST (see #184 ).  We only have a CST today, and a rather fragile one.

Before we can deliver an AST, we need a preprocessor (preferably integrated, but could live with it being standalone), see #183 .

Both semantic analysis and elaboration require domain expertise which I(we?) lack, but should everything before that come to fruition (preprocessor, parser, AST) someday, we'd like to produce a quality front-end so that no one ever has to re-invent yet-another SV parser ever again.

And "AST" could really be any SV interchange representation (pre-elaboration).  Getting the larger community to converge on this interchange form could be a challenge too.