chipsalliance/verible #378: Make macro-name-style lint rule parameterizable

msfschaffner, 2020-08-03:
It would be great if 

https://google.github.io/verible/lint.html#macro-name-style

could be configured in a similar way as

https://google.github.io/verible/lint.html#explicit-parameter-storage-type

In our case we would use this parameterizability to allow `lower_case` names for UVM macros.

Related: https://github.com/google/verible/issues/133

CC @sriyer

fangism, 2020-08-04:
The rule only covers macro definitions, but not call sites, so unless you're style-linting the UVM source code (or defining your own UVM-like macros), are you seeing false positive findings with this rule?

fangism, 2020-08-04:
I do generally agree that most of these *name-style-rules could be made configurable.

sriyerg, 2020-08-04:
We are allowing lower-case naming for macros that are very UVM adjacent:
https://github.com/lowRISC/opentitan/blob/a13393a3cdf2adc6151654fda4f08e9c2a07ec89/hw/dv/sv/dv_utils/dv_macros.svh#L6

I believe this is the same in our CDK area as well?

fangism, 2020-08-04:
Ok, those kind of look familiar.  
Since they are very localized to a central header file, would you be comfortable waiving a few affected ranges in that file?  or are there a lot more of such cases?

fangism, 2020-08-04:
I would also approve of ignoring all of `uvm.*` in general, which would leave only a few of your definitions in that file affected.  

(I'll run it by some folks to see if everyone generally agrees about that being uncontroversial.)

fangism, 2020-08-04:
I have a solution in the pipeline for UVM-macros, which get their own treatment out-of-the-box.  

Aside from that, this rule and others like it would probably benefit from sharing the same configuration options: enumerating the few common styles of names (off the top of my head):

* UpperCamelCase
* lower_snake_case
* ALL_CAPS

I'm open to proposals about the configuration details.

sriyerg, 2020-08-04:
Right, for enumeration as well, we allow both, ALL_CAPS and CamelCase. Please see:
https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md
https://github.com/lowRISC/style-guides/blob/master/DVCodingStyle.md

fangism, 2020-08-04:
Interesting, that means we'll need to be able to express OR, not just mutually exclusively one style.

Other variants from other styles:

* kUpperCamelCase

fangism, 2020-08-04:
Allowing for more than one probably means a user will want to make sure there is consistency within each file.
The way I've seen this done is that the first accepted style encountered becomes the style followed later in the file.

The style linter has no cross-file visibility, as with other similar style-checking tools.

fangism, 2020-08-04:
Looks like this can be modeled after 
https://github.com/google/verible/blob/master/verilog/analysis/checkers/parameter_name_style_rule.cc

@hzeller would you like to take this?  You may have had some refactoring in mind before copying that code over.

fangism, 2020-08-04:
b/162873562

sriyerg, 2020-08-04:
> The way I've seen this done is that the first accepted style encountered becomes the style followed later in the file.

Both styles can co-exist in the same file (consistency check within the same file is not needed): 
https://github.com/lowRISC/opentitan/blob/master/hw/vendor/lowrisc_ibex/rtl/ibex_pkg.sv

sriyerg, 2020-08-04:
> Interesting, that means we'll need to be able to express OR, not just mutually exclusively one style.
> 
> Other variants from other styles:
> 
> * kUpperCamelCase

We have not mandated kCamelCase - this is ci2 specific.

fangism, 2020-08-04:
> Both styles can co-exist in the same file (consistency check within the same file is not needed):
> https://github.com/lowRISC/opentitan/blob/master/hw/vendor/lowrisc_ibex/rtl/ibex_pkg.sv

Ok, no consistency checking then.

fangism, 2020-08-04:
> > Interesting, that means we'll need to be able to express OR, not just mutually exclusively one style.
> > Other variants from other styles:
> > 
> > * kUpperCamelCase
> 
> We have not mandated kCamelCase - this is ci2 specific.

Right, this is coming from internal style guidance.

karerua, 2022-02-03:
It would be great if all name-style rules have parameter to choose from 

lower_snake_case
CamelCase
kCamelCase
ALL_CAPS

It is very restrictive to have only one type of naming convention for different types