chipsalliance/verible #379: Set up sphinx + doxygen documentation generation

mithro, 2020-08-04:
We should publish verible documentation on [ReadTheDocs using Sphinx](https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html).

For HDL related projects like [google/skywater-pdk](https://github.com/google/skywater-pdk), [prjxray](https://github.com/SymbiFlow/prjxray) and [verilog-to-routing](https://github.com/verilog-to-routing/vtr-verilog-to-routing) we have standardized on using Sphinx (and on C++ projects Doxygen stuff being pulled into Sphinx).

There is also [sphinxcontrib-verilog-diagrams](https://github.com/SymbiFlow/sphinxcontrib-verilog-diagrams) which renders nice diagrams from Verilog and [sphinx_symbiflow_theme](https://github.com/SymbiFlow/sphinx_symbiflow_theme) which renders the documentation in material design.

mithro, 2020-08-04:
Related issues;
 - https://github.com/alainmarcel/Surelog/issues/305
 - https://github.com/alainmarcel/UHDM/issues/40

Demo documents;
 * The capnproto docs are done with sphinx -> http://capnproto.github.io/
 * The fomu workshop is done with sphinx -> http://fomu-workshop.fomu.im/
 * The SymbiFlow docs are done with sphinx -> https://symbiflow.readthedocs.io/en/latest/

mithro, 2020-09-28:
FYI -- @rw1nkler

mithro, 2020-09-28:
@rw1nkler set this up for the [VerilogToRouting repository](https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/master/doc/_doxygen) and it can be seen at https://docs.verilogtorouting.org/en/latest/api/vpr/contexts/

rw1nkler, 2020-09-29:
Some useful resources for adding Sphinx+Breathe integration (C/C++ support):
- [Breathe Quickstart](https://breathe.readthedocs.io/en/latest/quickstart.html)
- [Initial Sphinx+Breathe Support (VTR)](https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/1354/files)
- [API Documentation for C/C++ Projects (VTR)](https://docs.verilogtorouting.org/en/latest/dev/c_api_doc/)