chipsalliance/verible #382: forbid negative constant integer literals in declared array dimensions

fangism, 2020-08-06:
e.g. `logic [-1:0] foo;`

Suggestion:

* Create test cases with arrays in various places: module instances, ports, class members, local variables...
* Examine where they appear in `verible-verilog-syntax --printtree`
* Implement a `SyntaxTreeLintRule` to find offending locations.
* As needed use/write CST helper functions to extract substructure.
* Only focus on expressions that are constant literals.

fangism, 2020-08-06:
b/160002985

IEncinas10, 2023-03-12:
Hello, I would like to try solving this one.

I have made a bit of progress (I'm able to build, made the .h, .cc, test, I'm able to run them) but it looks like negative numbers are not detected by  https://github.com/chipsalliance/verible/blob/a1534abbd83b9ca5b84bc577e303724a9ebb5994/verilog/CST/expression.cc#L65

I guess the problem is already stated in the function declaration.

```c++
// Unwrap layers of only-child nodes until reaching a leaf or a node with
// multiple children.
const Symbol* DescendThroughSingletons(const Symbol& symbol);

```

GetDimensionRange[Left/Right]Bound gives us the node of each bound, but if we add a negative sign the node has a *kUnaryPrefixExpression*, that makes *DescendThroughSingletons* give up and return false (it detects that it has two childs).

```
Node @0 (tag: kUnaryPrefixExpression) {
  Leaf @0 (#'-' @37-38: "-")
  Node @1 (tag: kNumber) {
    Leaf @0 (#TK_DecNumber @38-39: "1")
  }
}
```

I guess the fix would involve being able to evaluate *kUnaryPrefixExpressions*. Should I try to add that functionality? Any pointers would be greatly appreciated!

On a side note, just to check whether I'm on the right path: I have basically inspired the new rule on the [unpacked dimensions rule](https://github.com/chipsalliance/verible/blob/master/verilog/analysis/checkers/unpacked_dimensions_rule.cc) (and packed one) so that we check *kDimensionRange*S inside both Packed and UnpackedDimensions.

```c++
static const Matcher& DimensionValueMatcher() {
  static const Matcher matcher(NodekDimensionRange());
  return matcher;
}

void ForbidNegativeArrayDim::HandleSymbol(
    const verible::Symbol& symbol, const verible::SyntaxTreeContext& context) {
  if (!ContextIsInsidePackedDimensions(context) && !ContextIsInsideUnpackedDimensions(context)) return;

  /* ... */
```

IEncinas10, 2023-03-13:
Hello! Thanks for the pointers. It seems I was a bit more off than expected. I couldn't get a proper compile_commands.json (followed [this](https://github.com/hedronvision/bazel-compile-commands-extractor), but it isn't 100% working) so I couldn't navigate too well the codebase.

I've narrowed the rule to match against *NodekUnaryPrefixExpression* inside packed/unpacked (same as before) as you suggested (thank you!). I got it working by writing ugly code that works by:

- Explicitly comparing leaf node against "-" string
- Checking if the child node is a ConstantIntegerValue

My (very simple? tests are now passing:

```c++
  const std::initializer_list<LintTestCase> kTestCases = {
      {""},
      {"logic l [",
       {kScalar, "-10"},
       ":0];"},
      {"logic [",
       {kScalar, "-10"},
       ":-0] l;"},
      {"logic l [+1:0];"},
      {"logic l [p:0];"},
      {"logic l [p:+0];"},
  };

```

But I wonder whether there is a nicer way to do it (functions already in place for processing UnaryPrefixExpressions?) The code is still very very hacky but just to give you a reference:

```c++
static const Matcher& UnaryPrefixExprMatcher() {
  static const Matcher matcher(NodekUnaryPrefixExpression());
  return matcher;
}

void ForbidNegativeArrayDim::HandleSymbol(
    const verible::Symbol& symbol, const verible::SyntaxTreeContext& context) {
  if (!ContextIsInsidePackedDimensions(context) && !ContextIsInsideUnpackedDimensions(context)) return;

  verible::matcher::BoundSymbolManager manager;
  if (UnaryPrefixExprMatcher().Matches(symbol, &manager)) {

    const auto& node = SymbolCastToNode(symbol);
    const auto& children = node.children();
    const auto& leaf_symbol = *children[0].get();
    const auto& term = verible::down_cast<const verible::SyntaxTreeLeaf&>(leaf_symbol);

    // mandatory? {leaf, node}
    if(children.size() <= 1) return;

    int value = -1;
    const auto &child_node = children[1];
    const bool is_constant = ConstantIntegerValue(*child_node.get(), &value);

    const verible::TokenInfo token(TK_OTHER, verible::StringSpanOfSymbol(symbol));
    if (is_constant && value > 0 && term.get().text() == "-") {
      violations_.insert(verible::LintViolation(token, kMessage, context));
    }
  }
}
```

Just wanted to check whether I got in the right direction so that I can focus on cleaning up the mess and making some more tests. 

Again, thank you for the pointers and the quick response

closed 2023-04-15 by hzeller