chipsalliance/verible #383: Recommend parentheses around certain expression patterns

fangism, 2020-08-06:
Though parsers and compilers know fully well the precedence of operators, humans can still benefit from extra parenthesis.

example:
https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#parentheses

Task: detect such situations (starting with the above ternary expression example) and report them

Suggest: subclassing a SyntaxTreeLintRule and for various expression types, looking downwards at its operands and comparing operators/precedences.  

Feel free to suggest more specific operator classes and cases.

fangism, 2020-08-06:
b/159846534

omarahmed1111, 2020-10-17:
@fangism Hello, I would like to work on this, could you assign me? Thanks

fangism, 2020-10-17:
Thank you!.  I have some lint-rule development documentation here:
https://github.com/google/verible/blob/master/doc/style_lint.md

I suggest starting by proposing a concrete set of test cases, given the initial description.
Feel free to propose your solution for discussion.  We're here to help.

fangism, 2020-10-27:
Thanks for working on this again.
I suggest we name this rule `"suggest-parentheses"` and the class `SuggestParenthesisRule`, and the topic anchor is named `"parentheses"`.

fangism, 2020-12-18:
Just checking in, what's the current status of this PR?

omarahmed1111, 2020-12-18:
Just need to add some tests for the helper functions and I will update it hopefully this week. Apologies for late updates as i have some issues the past weeks.

commit ecf05cba (2021-01-06): PR #583: Add suggest_parentheses rule for Verible::Linter
This PR introduces:
* Implementation of suggest_parentheses rule for the linter.
* Change kTernaryExpression to kConditionExpression globally.

suggest-parentheses rule is a new rule for the linter to suggest parentheses around nested expressions. The rule is not enabled by-default. Now, the rule only suggest parentheses around nested kConditionExpression in kConditionExpression.

Fixes #383

GitHub PR https://github.com/google/verible/pull/583

Copybara import of the project:

  - 4b76d67152d6dffd4fed4fa5343637a045ab448d Add the implem [... cut]
commit 51f6fc2f (2021-01-06): Enable [suggest-parentheses] style lint rule by default.

fixes #383

PiperOrigin-RevId: 350200313
closed 2021-01-06 by fangism