chipsalliance/verible #384: forbid line continuations inside string-literals

fangism, 2020-08-06:
The jury has spoken.  
We now want to forbid line-continued string literals like this:

```systemverilog
"this is bad. \
don't to this"
```

and recommend the equivalent (C-style joined string literals):

```systemverilog
"this is better. "
"do this instead."
```

Suggestion: implement as a `TokenStreamLintRule`, scan for all string literals (even inside macro definition bodies), and report offending locations (focusing on the location of "\").

fangism, 2020-08-06:
b/154548140

pawelsag, 2021-02-09:
Hi, I wanted to solve this issue, but I have a question related to the example you wrote.
I can't find any information about C-style string in TRM, so this is some kind of SystemVerilog dialect?
Does this also mean that when doing this task the implementations of the parser should be changed?
I prepared the example:
```
module strings; 
string tmp = "Humpty Dumpty sat on a wall."
"Humpty Dumpty had a great fall.";
endmodule
```
and the output was
```
build/strings.sv:4:1: syntax error, rejected ""Humpty Dumpty had a great fall."" (syntax-error).
```

snsokolov, 2021-02-09:
The C-style string joining above isn't a legal Verilog.  Strings can be concatenated by either by using `{}` or `$sformatf`

```
string tmp = {
    "foo",
    "bar"
};

string tmp2 = $sformatf("%s%s", "foo", "bar");
```

pawelsag, 2021-02-09:
Thanks for the clarification

PR #674 (2021-02-11): forbid line continuations inside string-literals
Fixes:  #384
commit 9d63188a (2021-02-11): forbid line continuations inside string-literals (#674)

* forbid line continuations inside string-literals
Adding:
 - TokenStreamLintRule definition for linter
 - tests for TokenStreamLintRule

Fixes #384
closed 2021-02-11 by hzeller