chipsalliance/verible #409: Backslash before newline in comment triggers indentation

hzeller, 2020-08-19:
Example:
```
bazel run //verilog/tools/formatter:verible-verilog-format -- - <<EOF
// ascii  \\ art
// sloping \\
// indented after backslash newline
// in subsequent comments.
EOF
```

Output:
```
// ascii  \ art
// sloping \
    // indented after backslash newline
    // in subsequent comments.
```

Debug partition tree output with `--show_token_partition_tree`
```
Full token partition tree:
{ ([<auto>], policy: always-expand) @{}
  { ([// ascii  \ art], policy: uninitialized) }
  { ([// sloping  \ // indented after backslash newline // in subsequent comments.], policy: fit-else-expand) }
}
```

fangism, 2020-08-19:
I am fairly certain this is the fault of the "InterLeaf" state machine that sweeps up non-syntax tokens in the Verilog tree_unwrapper.cc.  A recent change in the lexer separated out line continuations as distinct tokens (from the preceding //comment), and that state machine probably just needs to be updated to handle that.

cathales, 2023-12-19:
Hello, here is a similar issue:

```systemverilog
module something ();

  always_comb begin
    //aaaaaaaaaaaaaa\
    //aaaaaaaaaaaaaaaaaaaaaaaaa
    for (int unsigned i = 0; i < TLB_ENTRIES; i++) begin
      $fatal();
    end
  end

endmodule
```

(comments with a backslash)

becomes

```systemverilog
module something ();

  always_comb begin
    //aaaaaaaaaaaaaa\
    //aaaaaaaaaaaaaaaaaaaaaaaaa
    for (
        int unsigned i = 0; i < TLB_ENTRIES; i++
    ) begin
      $fatal();
    end
  end

endmodule
```

(The `for` line is splitted but it should not.)

---

```systemverilog
module something ();

  always_comb begin
    for (int unsigned i = 0; i < TLB_ENTRIES; i++) begin
      $fatal();
    end
  end

endmodule
```

remains

```systemverilog
module something ();

  always_comb begin
    for (int unsigned i = 0; i < TLB_ENTRIES; i++) begin
      $fatal();
    end
  end

endmodule
```

---

When it encounters a `\` at the end of a comment, it seems to cumulate the number of characters on all the comment lines until the code line included, including the carriage return bytes between the lines. When it exceeds 110 then it seems to consider that it is too long so it "cuts" the `for` line.

In the example above, if I remove an `a` from one of the comment lines, then the `for` line is not splitted.

kbrunham-intel, 2026-09-22:
@hzeller I retested the reduced case from this issue with `verible-verilog-format` built from tag `v0.0-4294-gc1d8f5e8`. I could not reproduce the problem (formatter exits 0; output keeps the expected layout / converges). I would suggest closing this issue.

The original `// sloping \\` comment repro no longer over-indents the following comment lines. (Related leftover: #2243 ASCII-art inside a module still extra-indents the line after a trailing `\`.)