chipsalliance/verible #420: Kythe: extract module instance parameters as children not as anchors

MinaToma, 2020-09-02:
Extraction of " module_type m(a,b,c,d)" needs to change as a,b,c,d are now anchors and should be considered as children to be able to extract "module_type m(id1[id2]);"

MinaToma, 2020-09-16:
@fangism I tried this example and it worked fine (I think using `wire` as an index isn't right but ignore it as `int` isn't extracted yet)

```
module simple(input a, output b);
endmodule

module with_array();

    wire a[1:0];
    wire x = 0, y = 1;

    simple sim(a[x], a[y]);

endmodule
```

MinaToma, 2020-09-16:
Can the port be class member or some package member?

closed 2020-09-16 by hzeller

fangism, 2020-09-16:
> @fangism I tried this example and it worked fine (I think using `wire` as an index isn't right but ignore it as `int` isn't extracted yet)
> 
> ```
> module simple(input a, output b);
> endmodule
> 
> module with_array();
> 
>     wire a[1:0];
>     wire x = 0, y = 1;
> 
>     simple sim(a[x], a[y]);
> 
> endmodule
> ```

That's ok, it should be able to tolerate what would otherwise fail semantically.  Source code validation is the responsibility of other tools.

fangism, 2020-09-16:
> Can the port be class member or some package member?

I think so, if they are compile-time constant values.

commit fe94fa5b (2020-09-16): PR #458: Extracting ActualPorts and Module Instance Ports
Fixes #420
WIP #394
WIP #434

GitHub PR https://github.com/google/verible/pull/458

Copybara import of the project:

  - 2cbba9f96e5b897a0dd653f1bc462a5c83913c87 Extracting ActualPorts and Module Instance Ports by MinaToma <minatoma1999@gmail.com>
  - 75571b8d0fe9a9d7b0e1c4b2d07b3ce278ac2fe1 Refactoring by MinaToma <minatoma1999@gmail.com>
  - ce3ec3f1413516e646105247e1a60329604dead5 Fix Typo by MinaToma <minatoma1999@gmail.com>
  - dd6664b4b12e9c51047c7ea9e49bda0122e5ce56 Merge ce3ec3f1413516e646105247e1a60329604dead5 into 29abb... by  [... cut]