chipsalliance/verible #433: Align function/task formal parameters

fangism, 2020-09-11:
Example from: https://github.com/msfschaffner/opentitan/pull/7/files/2bf9354bef93019b3778564ea746aa51584e8d2d#diff-f7facda2840d21c08bab1b5280a8d485

Desirable output:

```systemverilog
  import "DPI-C" context function void c_dpi_aes_crypt_message(
    input  bit              impl_i,
    input  bit              op_i,
    input  bit        [5:0] mode_i,
    input  bit  [3:0][31:0] iv_i,
    input  bit        [2:0] key_len_i,
    input  bit  [7:0][31:0] key_i,
    input  bit        [7:0] data_i[],
    output bit        [7:0] data_o[]
  );
```

However, I sense that forcing parameters onto their own line may not always be desirable.  We'll need to come up with a way to sense the author's intent.

wtatarski, 2021-05-14:
@fangism  @msfschaffner

I have some doubts about aligning two dimensional arrays in function declaration. Shouldn't it be in that way? 
```
  import "DPI-C" context function void c_dpi_aes_crypt_message(
    input  bit             impl_i,
    input  bit             op_i,
    input  bit [5:0]       mode_i,
    input  bit [3:0][31:0] iv_i,
    input  bit [2:0]       key_len_i,
    input  bit [7:0][31:0] key_i,
    input  bit [7:0]       data_i   [],
    output bit [7:0]       data_o   []
  );
```

Actually this is how verible aligns dimensions in module declaration.

fangism, 2021-05-14:
The multi-dimension formatting is really up to user preference which is ... mixed.  Ideally, it should be style-controlled per project.

msfschaffner, 2021-05-14:
Indeed, I have seen both flush-left and flush-right alignment of multi-dimensional arrays.
There is no right or wrong, and configurability would be great.

PR #819 (2021-05-28): Align parameters in exported/imported function
Fixes #433
closed 2021-05-28 by tgorochowik