chipsalliance/verible #444: Missing space before hashtag in class declarations extending parameterized class

msfschaffner, 2020-09-11:
The formatter removes the space before the hashtag:

before:
```systemverilog
class aes_env_cfg extends cip_base_env_cfg #(.RAL_T(aes_reg_block));
```
after:
```systemverilog
class aes_env_cfg extends cip_base_env_cfg#(
    .RAL_T(aes_reg_block)
);
```

msfschaffner, 2020-09-11:
b/168309966

vasantteja, 2020-09-15:
Can I assign this issue to myself?

fangism, 2020-09-15:
Yes, thank you for volunteering.

I expect the fix for this to be a few lines in verilog/formatting/token_annotator.cc.

Tip:
If you run `verible-verilog-format` with `-v=3 --alsologtostderr`, it will show you the reason why a particular spacing value was chosen between two tokens.

fangism, 2020-09-22:
@vasantteja Checking in, need any help? Have any questions?

vasantteja, 2020-09-30:
@fangism sorry, I was traveling and I couldn't take a look at this. I will reach out to you if I have any doubts. Thanks for the patience.

PR #705 (2021-03-23): formatter: Fix missing space before hashtag
Fix missing space before hashtag in class declarations extending parameterized class:
```
class aes_env_cfg extends cip_base_env_cfg #(
    .RAL_T(aes_reg_block)
);
```

Fixes #444
commit aad170a5 (2021-03-23): Fixe missing space before hashtag (pull-request #705)

formatter: Fix missing space before hashtag
Fixes #444
closed 2021-03-23 by hzeller