chipsalliance/verible #459: [kythe] crash: CHECK failure: !stack_.empty() in auto_pop_stack.h

fangism, 2020-09-16:
## Input test case

(not reduced, direct from fuzz-testing report)

[clusterfuzz-testcase-minimized-verilog_kythe_extractor_fuzzer-6216524817498112.txt](https://github.com/google/verible/files/5234347/clusterfuzz-testcase-minimized-verilog_kythe_extractor_fuzzer-6216524817498112.txt)

## Describe what is wrong or missing

```
Crash Type: CHECK failure
Crash Address:
Crash State:
  !stack_.empty() in auto_pop_stack.h
  verible::AutoPopStack<verible::VectorTree<verilog::kythe::IndexingNodeData>*>::t
  verilog::kythe::IndexingFactsTreeExtractor::ExtractModuleInstantiation
```

I think this is automatically entering a parsing mode where the source code is interpreted as a module-body, and this probably breaks some fundamental assumptions.  Such files are typically `` `include``d by other files and make sense in those context, but not standalone.  Either way, we should just let the extractor gracefully exit.

fangism, 2020-09-16:
b/168722009

MinaToma, 2020-09-30:
Are we still having problems with this test?
I don't understand the test can you simplify it?

fangism, 2020-09-30:
No longer crashes after the above commit, and just results in syntax errors, so working as intended now.

closed 2020-09-30 by fangism