chipsalliance/verible #460: Crash on $size in for loop

zegervdv, 2020-09-17:
**Test case**

```systemverilog
module foo ();
always_comb begin
  for (int i = 0; i < $size(s_req); i++) begin
    s_req[i] = 1;
  end
end
endmodule
```

**Actual output**

Formatter crashes.

Include any possible diagnostic messages from the formatter.
```
F common/formatting/token_partition_tree.cc:62] Check failed: parent_end == children_end (14 vs. 13)node:
{ (>>>>>>>>[<auto>], policy: fit-else-expand) @{1,1,11,1,6,0,1,1}, (origin: "i < $size(s_req)")
{ (>>>>>>>>[i < $size (], policy: fit-else-expand) }
{ (>>>>>>>>>>>>[s_req], policy: fit-else-expand, (origin: "s_req")) }
{ (>>>>>>>>[)], policy: uninitialized) }
```

The error can be circumvented by assigning the `$size(s_req)` to a constant first and using the constant in the for loop.

fangism, 2020-09-17:
Confirmed, able to reproduce here.  Thanks for the report.

fangism, 2020-09-17:
b/168826199

fangism, 2020-09-22:
@hzeller You found a very similar case to this one with `$bits()` inside a for-loop.  Do you have full context to paste?

fangism, 2020-09-26:
Running the original test case with `-v=4 --alsologtostderr`, we see:

```
I0925 16:52:53.596272 2913354 tree_unwrapper.cc:1328] ReshapeTokenPartitions node: kForSpec
I0925 16:52:53.596274 2913354 tree_unwrapper.cc:1330] before reshaping kForSpec:
{ (>>>>>>>>[int i = 0 ; i < $size ( s_req ) ; i ++], policy: fit-else-expand, (origin: "int i = 0; ...s_req); i++"))
  { (>>>>>>>>[int i = 0], policy: fit-else-expand, (origin: "int i = 0")) }
  { (>>>>>>>>[;], policy: uninitialized) }
  { (>>>>>>>>[i < $size ( s_req )], policy: fit-else-expand, (origin: "i < $size(s_req)"))
    { (>>>>>>>>[i < $size (], policy: fit-else-expand) }
    { (>>>>>>>>>>>>[s_req], policy: fit-else-expand, (origin: "s_req")) }
    { (>>>>>>>>[)], policy: uninitialized) }
  }
  { (>>>>>>>>[;], policy: uninitialized) }
  { (>>>>>>>>[i ++], policy: fit-else-expand, (origin: "i++")) }
}
I0925 16:52:53.596285 2913354 tree_unwrapper.cc:1678] kForSpec got ';' at 1 and 3
I0925 16:52:53.596289 2913354 tree_unwrapper.cc:1781] after reshaping kForSpec:
{ (>>>>>>>>[int i = 0 ; i < $size ( s_req ) ; i ++], policy: fit-else-expand, (origin: "int i = 0; ...s_req); i++"))
  { (>>>>>>>>[int i = 0 ;], policy: fit-else-expand, (origin: "int i = 0")) }
  { (>>>>>>>>[i < $size ( s_req ) ;], policy: fit-else-expand, (origin: "i < $size(s_req)"))
    { (>>>>>>>>[i < $size (], policy: fit-else-expand) }
    { (>>>>>>>>>>>>[s_req], policy: fit-else-expand, (origin: "s_req")) }
    { (>>>>>>>>[)], policy: uninitialized) }
  }
  { (>>>>>>>>[i ++], policy: fit-else-expand, (origin: "i++")) }
}
I0925 16:52:53.596297 2913354 tree_unwrapper.cc:1782] end of ReshapeTokenPartitions node: kForSpec
```

The `;` in `...; i++` got cut out.

fangism, 2020-09-26:
This crashes with nearly the same signature.

```systemverilog
module foo ();
always_comb
  for (int i = $size(n); i >=0; i--) begin
  end
endmodule
```

I've figured out the fix.

commit 6b0edca5 (2020-09-29): Use safer reshape function for merging ';' into predecessor.

fixes #460

PiperOrigin-RevId: 334029119
closed 2020-09-29 by hzeller