chipsalliance/verible #474: [kythe] Extend Kythe Extractor to cover `include for Cross-File Referencing

MinaToma, 2020-09-22:
## Describe what is wrong or missing

Extend the extractor to extract facts for ```include`` and cover cross-file referencing.

fangism, 2020-09-22:
You may assume that given a file list, the files named are top-level files from which to start, and that included files referenced are accessible on the filesystem relative to the path from which the tool is invoked.  Later we can add an option to support include-paths (like `-I` for C/C++ compilers) that will add flexibility.

fangism, 2020-09-22:
Possible approaches to this problem:

* Textually expand the included file inline like a preprocessor would, into a single text stream.  But you'd then need to keep track of originating text locations in the file stack (in case of nested includes).  This has a better chance of being able to parse the expanded result in cases where the included files are not parseable on their own.  The downside is that preprocessing expansion has not been built out yet.
* Do not expand, but parse the included file(s) on their own (taking advantage of the automatic alternate parsing mode that can handle some included files' syntax) to yield a CST and retain a file stack.  "Enter" the CST for the included file after it has been parsed in the same traversal that started the outer/includer file.  In the uncommon but plausible event that a file is included from two places, you could cache a CST per unique file, and save duplicate parsing work.  Effectively, this gives the _appearance_ of a single CST that is synthesized from multiple files.

MinaToma, 2020-09-25:
Closing this issue as it seems duplicated with this one #489.
But thanks for the suggestions.

closed 2020-09-25 by MinaToma