chipsalliance/verible #480: [kythe] `super` should reference extends class where applicable

fangism, 2020-09-22:
## Input test case

```systemverilog
class core_ibex_base_test extends uvm_test;
  virtual function void build_phase(uvm_phase phase);
    super.build_phase(phase);
  endfunction
endclass
```

## Describe what is wrong or missing

`super` should reference the `uvm_test` even, if its definition is unavailable.
(It comes from a standard library called UVM, analogous to the C++ standard library.)

Self-contained reduced test case:

```systemverilog
class base_c;
  virtual function void f();
  endfunction
endclass

class derived_c extends base_c;
  virtual function void f();
    super.f();  // super -> base_c, and f -> base_c::f
  endfunction
endclass
```

If there is more than one extends class (uncommon), you can give up, or take a best guess, or try to infer the correct base class reference.

fangism, 2020-09-22:
Currently, the `f` in `super.f()` points to self `derived_c::f`.

MinaToma, 2020-09-25:
Related to #436

fangism, 2020-11-10:
Clarification: extends can only take one class, so `super` should resolve to that.
(Interface classes #535 are different.)