chipsalliance/verible #484: [kythe] pre-index UVM library

fangism, 2020-09-22:
This is a bit of an open-ended request to somehow incorporate indexing of the standard UVM library so that projects that use it can link to an authoritative source of information.  The UVM library rarely changes, and is heavily used, so there would be significant benefit at a (roughly) one-time cost to index it serve it.

Example repo: https://github.com/accellera/uvm

Also note: there have been a few released versions of UVM over the years.

Some research needed to learn what other languages with standard libraries do, and what best practices around cross-repo references are.

fangism, 2020-10-20:
@mithro @ivan444 @hzeller How do you think we should go about this task?  This is a "standard library" whose source code is available, and projects are not expected to have to include a copy of it.  I noticed in Verible's source tree that external libraries like absl appear under `external`, [example](https://cs.opensource.google/verible/verible/+/master:external/com_google_absl/absl/strings/string_view.h;drc=master;l=0).

I suspect this is being re-indexed by Verible's regular releases, so it's not actually linking to any pre-indexing of the absl library upstream.  

There are no build rules for Verilog yet, so Verilog indexing may have to be done via scripts.  Maybe the process could emulate what bazel rules do for C++, by staging dependent libraries in an `external` area (with a UVM checkout like a git submodule), and have our multi-file kythe indexer pass that path as an `-incdir`, including the UVM package files in the top-level filelist.

ivan444, 2020-10-20:
This is a question of deployment. I'm all for documenting how to index UVM and make it work along your codebase, but offering a preindexed UVM might be a premature optimization.

We need to make sure that cross corpus references work well (i.e., Kythe VNames are well constructed when crossing the project boundry).