chipsalliance/verible #491: transition() is experimental and disabled by default

mithro, 2020-09-22:
```
ERROR: /tmpfs/tmp/bazel/external/com_github_google_rules_install/installer/def.bzl:87:32: transition() is experimental and disabled by default. This API is in development and subject to change at any time. Use --experimental_starlark_config_transitions to use this experimental API.
ERROR: /tmpfs/tmp/bazel/external/com_github_google_rules_install/installer/def.bzl:100:19: Traceback (most recent call last):
	File "/tmpfs/tmp/bazel/external/com_github_google_rules_install/installer/def.bzl", line 93
		rule(implementation = _gen_install_bi..., <3 more arguments>)
	File "/tmpfs/tmp/bazel/external/com_github_google_rules_install/installer/def.bzl", line 97, in rule
		attr.label_list(mandatory = True, allow_files = Tr..., ...)
	File "/tmpfs/tmp/bazel/external/com_github_google_rules_install/installer/def.bzl", line 100, in attr.label_list
		_compilation_mode_transition
global variable '_compilation_mode_transition' is referenced before assignment.
ERROR: error loading package '': Extension file 'installer/def.bzl' has errors
```

hzeller, 2020-09-23:
When do you see this message ?

hzeller, 2020-09-23:
What bazel version are you using ?

mithro, 2020-09-23:
This is from the Kokoro Kythe builder internally.