chipsalliance/verible #492: warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]

mithro, 2020-09-22:
```
tansell@tansell:~/tmp/verible$ bazel --bazelrc=/opt/kythe-v0.0.38/extractors.bazelrc build --override_repository kythe_release=/opt/kythe-v0.0.38 --define=kythe_corpus=github.com/google/verible -- //...
<snip>
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis/checkers:module_filename_rule_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis:verilog_analyzer:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis/checkers:endif_comment_rule_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/CST:dimensions_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/CST:declaration_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/CST:functions_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/CST:match_test_utils:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/tools/lint:verible-verilog-lint:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/CST:package_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/CST:port_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis:verilog_linter_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis:verilog_linter:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis/checkers:struct_union_name_style_rule_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis/checkers:explicit_task_lifetime_rule_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis/checkers:posix_eof_rule_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/parser:verilog_parser_unittest:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis/checkers:signal_name_style_rule_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis:verilog_linter_configuration_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis/checkers:constraint_name_style_rule_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/formatting:comment_controls:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/tools/diff:verible-verilog-diff:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis/checkers:module_instantiation_rules_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis:verilog_equivalence:
[... 8 lines cut ...]
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/transform:obfuscate_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/analysis:verilog_equivalence_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/tools/obfuscator:verible-verilog-obfuscate:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/formatting:formatter_tuning_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/formatting:formatter:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/tools/formatter:verible-verilog-format:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: From Executing extra_action @kythe_release//:extract_kzip_cxx_extra_action on //verilog/formatting:formatter_test:
warning: unknown warning option '-Wno-free-nonheap-object'; did you mean '-Wno-sequence-point'? [-Wunknown-warning-option]
1 warning generated.
INFO: Elapsed time: 301.884s, Critical Path: 184.38s
INFO: 2147 processes: 2147 linux-sandbox.
INFO: Build completed successfully, 4513 total actions
```

fangism, 2020-09-23:
This is coming from kythe's own BUILD files, which probably assumes some range of versions of compilers.  (bazel != autoconf/automake)
If it's a nuisance, maybe we can patch it out?

mithro, 2020-09-23:
This is from the Kokoro Kythe builder internally.