chipsalliance/verible #493: want space between return keyword and following expression

fangism, 2020-09-23:
**Test case**

```systemverilog
  function int foo(logic [31:0] data);
    return {<<8{data}};
  endfunction
```

**Actual output**

```systemverilog
  function int foo(logic [31:0] data);
    return{<<8{data}};
  endfunction
```

**Expected or suggested output**

[should be the original spacing after `return`]

fangism, 2020-09-23:
b/169194747

ahmedkrmn, 2020-10-03:
Hi @fangism,
Can I take this one too?

fangism, 2020-10-03:
This is the function you'll want to update:
https://cs.opensource.google/verible/verible/+/master:verilog/formatting/token_annotator.cc;bpv=1;bpt=1;l=552?gsn=SpacesRequiredBetween&gs=kythe%3A%2F%2Fgithub.com%2Fgoogle%2Fverible%3Flang%3Dc%252B%252B%3Fpath%3Dverilog%2Fformatting%2Ftoken_annotator.cc%23ROmaSUMU_uyvJwD04Z9lzyIO9K0QZQn_6cJo7pfIUAg

DhairyaBahl, 2020-10-15:
@fangism I can't see any activity on this from few days. If its still available, Kindly assign it to me.

Thanks :)

ahmedkrmn, 2020-10-15:
Hi,
I've been busy lately, but I will start working on it this weekend.

DhairyaBahl, 2020-10-15:
@ahmedkrmn No offense bro. I was just finding some issues for me ;)

fangism, 2020-10-15:
Thanks, both of you for offering!

commit 93e149db (2020-10-21): PR #542: Add space between return keyword and return value
Fixes #493.

**Input:**
```verilog
  function int foo(logic [31:0] data);
    return {<<8{data}};
  endfunction
```
**Formatter output before:**
```verilog
  function int foo(logic [31:0] data);
    return{<<8{data}};
  endfunction
```

**Formatter output after:**
```verilog
  function int foo(logic [31:0] data);
    return {<<8{data}};
  endfunction
```

GitHub PR https://github.com/google/verible/pull/542

Copybara import of the project:

  - 9fceb038fa6669d815bb0f61925ab3f08032e948 Fix spacing between return keyword and return value [... cut]
closed 2020-10-21 by hzeller