chipsalliance/verible #494: Align colon inside dimension ranges

fangism, 2020-09-23:
**Test case**

```systemverilog
class c;
  rand bit [DataWidth - 1   : 0]   d_data;
  rand bit [SourceWidth - 1 : 0] d_source;
endclass
```

**Actual output**

```systemverilog
class c;
  rand bit [DataWidth - 1 : 0]   d_data;
  rand bit [SourceWidth - 1 : 0] d_source;
endclass
```

Include any possible diagnostic messages from the formatter.

**Expected or suggested output**

(want the original form)

fangism, 2020-09-23:
b/158244946

fangism, 2020-09-23:
This one may be a little tricker than just reserving new columns around the `:` because the contents of `[]` could be a scalar value, not always a range.

ldalek-antmicro, 2021-06-08:
More examples:
1. [hw/ip/sram_ctrl/dv/env/seq_lib/sram_ctrl_multiple_keys_vseq.sv
](https://github.com/lowRISC/opentitan/blob/master/hw/ip/sram_ctrl/dv/env/seq_lib/sram_ctrl_multiple_keys_vseq.sv#L13)
```diff
   // Use 2 as the low end since `num_trans == 1` is the default for the smoke test.
   constraint num_trans_c {
     num_trans dist {
-      [2  : 13] :/ 2,
+      [2 : 13]  :/ 2,
       [14 : 25] :/ 3,
       [25 : 37] :/ 4,
       [38 : 50] :/ 1
```
2. [hw/ip/sram_ctrl/dv/env/seq_lib/sram_ctrl_stress_pipeline_vseq.sv
](https://github.com/lowRISC/opentitan/blob/master/hw/ip/sram_ctrl/dv/env/seq_lib/sram_ctrl_stress_pipeline_vseq.sv#L14)
```diff
   // to prevent simulations taking forever
   constraint num_ops_c {
     num_ops dist {
-      [1 : 1000] :/ 1,
+      [1 : 1000]    :/ 1,
       [1001 : 2500] :/ 6,
       [2501 : 4000] :/ 2,
       [4001 : 5000] :/ 1
```

PR #865 (2021-09-10): Align colon inside ranges; Subcolumns support in tabular alignment
Two features in a single PR to avoid dependent PRs. Each commit is one feature.

* **Add VectorTree iterators**
  Adds code that allows to traverse VectorTree in specific order using range for loop. Supported traversals:
  * Leaves
  * Pre order
  * Post order

* **Add support for subcolumns in tabular alignment mode**
  Converts columns structure from a vector to a vector tree. This allows to use subcolumns (visual example below; example use in the following commit).

* **Align dimensions and ranges in class properties and dis [... cut]
closed 2021-09-10 by hzeller