chipsalliance/verible #498: [alignment] un-indented package import statement in FPV module declaration

udinator, 2020-09-23:
**Test case**

```systemverilog
module alert_handler_esc_timer_fpv import alert_pkg::*; (
  // i/o ports
);
```

**Actual output**

```systemverilog
module alert_handler_esc_timer_fpv
import alert_pkg::*;
(
  // i/o ports
);
```

**Expected or suggested output**

Ideally the `import` statement should be indented, e.g.

```systemverilog
module alert_handler_esc_timer_fpv
  import alert_pkg::*;
( 
  // i/o ports
);
```

fangism, 2020-09-23:
b/169252240

ahmedkrmn, 2020-10-01:
Hi @fangism,
I would like to take up this issue. Any guidance on how to implement this?

fangism, 2020-10-01:
Thanks.  Some general tips:

Run `verible-verilog-syntax --printtree` on examples of input code.  Note the names of the nodes that capture the sections of interest.  In this case, it should be `kPackageImportList` which was constructed here:

https://cs.opensource.google/verible/verible/+/master:verilog/parser/verilog.y;drc=d7cd6eae543434970bcc8d10f3dd2a19a8a6c2df;l=2903

In https://cs.opensource.google/verible/verible/+/master:verilog/formatting/tree_unwrapper.cc you'll want to handle the case for `kPackageImportList`.  Take a look at how other Lists are handled for example -- many of them start indented sections.  For `kPackageImportDeclaration`, I *think* you don't need to do anything because it is already handled.

Add/update tests in tree_unwrapper_test.cc for verifying the new internal structure with your change, and also formatter_test.cc for end-to-end examples.

Feel free to ping here with more questions.

fangism, 2020-10-01:
Another way to interactively test/see the token partition tree structure produced by tree_unwrapper.cc is to run `verible-verilog-format --show_token_partition_tree` on an input file.

commit 4eb091df (2020-10-07): PR #520: Fix module-level package import indentation
Fixes #498.

**Input:**
```verilog
module foo import bar::*; (baz); endmodule
```
**Formatter output before:**
```verilog
module foo
import bar::*;
(
    baz
);
endmodule
```

**Formatter output after:**
```verilog
module foo
  import bar::*;
(
    baz
);
endmodule
```

GitHub PR https://github.com/google/verible/pull/520

Copybara import of the project:

  - e8ab4f28bff5fe7d0d6014b1f79884c9d4777de1 Fix module-level package import indentation by Ahmed Karaman <ahmedkhaledkaraman@gmail.com>
  - dc63afd56085da1b7637f0ed253a181f7bcfd74d Merge e [... cut]
closed 2020-10-07 by fangism