chipsalliance/verible #499: [alignment] incorrect indentation of `initial begin...end` blocks

udinator, 2020-09-23:
**Test case**

```systemverilog
initial // clock generation
  begin
    clk = 0;
    forever begin
      #4ns clk = !clk;
    end
  end
```

**Actual output**

```systemverilog
initial // clock generation
    begin
  clk = 0;
  forever begin
    #4ns clk = !clk;
  end
end
```

**Expected or suggested output**

The formatter should not touch this input block at all as it is already formatted correctly.

```systemverilog
initial // clock generation
  begin
    clk = 0;
    forever begin
      #4ns clk = !clk;
    end
  end
```

Citations to published style guides would help.

fangism, 2020-09-23:
b/169252975

wtatarski, 2021-06-01:
@fangism  @udinator 
At this moment the actual output looks like this:
```
initial // clock generation
  begin
  clk = 0;
  forever begin
    #4ns clk = !clk;
  end
end
```
Do you think that additional indentation is required after `begin`?

udinator, 2021-06-01:
Hi @wtatarski - I'd prefer that we keep the additional indentation after the `begin`.

PR #857 (2021-06-16): Fix indenatation of initial begin with EOL comment
Fixes #499
closed 2021-06-16 by tgorochowik