chipsalliance/verible #519: Override style-guide citation links and references

fangism, 2020-10-01:
**Summary**

Currently, style guide citation links are hard-coded into the rule implementations themselves.
In the future, users and rule-set maintainers may wish to redirect these to point to their own style guides.

**Proposal**

Have a way at run-time to re-map each rule's citation to a new link, for example loading such a map through a command-line option that reads an external configuration.

fangism, 2020-10-01:
b/169867142

pawelsag, 2021-04-02:
I thinking about loading the custom style guide citation from the JSON file. What do you think about this idea?

mithro, 2021-04-02:
I would suggest that we follow whatever other tools do in this area? What do clang-format and pylint do?

pawelsag, 2021-04-23:
Most of the tools are using just plain text configuration in a format
```
key1: value
key2: value
```
I will do the same for our config:
```
rule 1: style guide citation 1
rule 2: style guide citation 2
```