chipsalliance/verible #521: [kythe] CHECK failure: GetSubtreeAsSymbol(symbol, parent_must_be_node_enum, child_position)

fangism, 2020-10-04:
## Input test case

Try to reduce the test down to a single demonstrative file.

```systemverilog
n;
```

## Describe what is wrong or missing

```
Crash Type: CHECK failure
Crash Address:
Crash State:
  GetSubtreeAsSymbol(symbol, parent_must_be_node_enum, child_position)' Must be no
  verible::Symbol const* absl::logging_internal::DieIfNull<verible::Symbol const*>
  verible::SyntaxTreeNode const& verible::GetSubtreeAsNode<verible::Symbol, verilo
```

MinaToma, 2020-10-23:
This one is not crashing any more but I found something else.
This is the CST for `n;`:

```
Node @0 (tag: kDescriptionList) {
  Node @0 (tag: kDataDeclaration) {
    Node @0 (tag: kQualifierList) {
    }
    Node @1 (tag: kInstantiationBase) {
      Node @0 (tag: kInstantiationType) {
        Node @0 (tag: kDataType) {
        }
      }
      Node @1 (tag: kVariableDeclarationAssignmentList) {
        Node @0 (tag: kVariableDeclarationAssignment) {
          Node @0 (tag: kUnqualifiedId) {
            Leaf @0 (#SymbolIdentifier @0-1: "n")
          }
          Node @1 (tag: kUnpackedDimensions) {
          }
        }
      }
    }
    Leaf @2 (#';' @1-2: ";")
  }
}
```

probably related to #549

MinaToma, 2020-10-23:
This is not crashing now.

closed 2020-10-23 by MinaToma