chipsalliance/verible #525: Check for legacy genvar declarations and generate region style

msfschaffner, 2020-10-06:
**Summary**

The google and [lowRISC style guides](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#generate-constructs) discourage the use of `generate` regions and separate `genvar` declarations, since they are considered legacy constructs.

It would be nice if the style linter could detect those constructs and flag them.

Suggested rule names:
`[legacy-genvar-declaration]`
`[legacy-generate-region]`

**Test cases and examples**

The more examples you provide, the easier it will be for someone to come along
and implement it.

bad:
```systemverilog
genvar k;
// may have code her as well
generate
  for (k = 0; k < FooParam; k++) begin : gen_loop
    // code
  end
endgenerate

generate
  if (FooParam) begin : gen_if
  // code
  end
endgenerate
```

good:
```systemverilog
for (genvar k = 0; k < FooParam; k++) begin : gen_loop
  // code
end

if (FooParam) begin : gen_if
  // code
end
```

msfschaffner, 2020-10-06:
b/170256350

fangism, 2020-10-07:
There is a closely related rule [v2001-generate-begin]:

https://cs.opensource.google/verible/verible/+/master:verilog/analysis/checkers/v2001_generate_begin_rule.h

That one checks explicitly for `generate begin`.  I should ask around if that one can be generalized to cover all uses of `generate`.

fangism, 2020-10-07:
In both places I checked, including the lowRISC guide section you cited, I see:

```
Do not wrap a generate construct with an additional begin block.

Do not use generate regions {generate, endgenerate}.
```

So they are in agreement.  

Of these statements, the second seems to cover the first, making it somewhat redundant.  WDYT?

If not redundant, should these two statements be implemented as one rule or separate?

msfschaffner, 2020-10-07:
Indeed, the second one seems to be a super set of the first with respect to style guidance (i.e., don't use generate regions). Hence I would be ok with just implementing one rule for this, unless we see a need that this has to be made configurable in the future (and in such a case you may still want both rules).

PR #649 (2021-01-27): Check for genvar and generate region
Closes #525

Note that this does not remove `v2001-generate-begin` rule mentioned in the issue.
closed 2021-01-27 by hzeller