chipsalliance/verible #528: [fun] tool/script to animate text formatting/transformation as a demo

fangism, 2020-10-09:
For fun, I'd like to create an animated demo of formatting (or any text transformation):

Script Input for producing text images:

* A batch of (unformatted) text files
* Any transformation command, like sed
* Some configuration of how to render text (font, size, foreground/background color) as an image

Input for animation:

* Configuration of duration for each image (before/after transformation)

Script Output:

* Intermediate images of said text files (could be SVG)
* GIF or similar animation of text images before and after (formatting) transformation

DhairyaBahl, 2020-10-15:
@fangism I would like to work on this issue.

Thanks :)

fangism, 2020-10-15:
Thank you!  Looking forward to learning how you do this.

DhairyaBahl, 2020-10-15:
@fangism Although I am new to this organisation, I will do my best.

Thanks :)

DhairyaBahl, 2020-10-17:
@fangism I am stuck a bit !! Is it really a good first issue ?? Kindly help.

Thanks :)

fangism, 2020-10-17:
For the first step, I'm looking for an open-source solution for converting [text to jpg or svg](https://www.google.com/search?q=text+file+to+image&oq=text+file+to+image&aqs=chrome..69i57j0l4j69i64l3.4712j0j1&sourceid=chrome&ie=UTF-8).  Those search results show online web-based converters, but some offline tool would be better.
I found ["PIL"](https://stackoverflow.com/questions/17856242/how-to-convert-a-string-to-an-image) and ["Pillow"](https://pillow.readthedocs.io/en/stable/) for Python, which could be interesting.

fangism, 2020-11-03:
Just checking in, had any time to look into this, @DhairyaBahl ?

DhairyaBahl, 2020-11-04:
@fangism I am extremely sorry for delay towards the issue. I will try my best to push a PR in a day or so. 

Thank you :) sorry again :)

DhairyaBahl, 2020-11-14:
@fangism Thank you for your help and effort with me. I look forward to work with this project again. I am unassigning myself from this issue as well.

fangism, 2020-11-14:
> @fangism Thank you for your help and effort with me. I look forward to work with this project again. I am unassigning myself from this issue as well.

Yes, we hope to see you back some day.

SeanMcLoughlin, 2021-02-20:
Hey, this sounds interesting. Although, I'm not entirely sure what the ask is here. Do you want a tool that can generate before/after formatting images, or generate a GIF of doing formatting, regardless of the input text?

Or do you simply want a GIF that demonstrates how before and after formatting looks on some basic file? I'd imagine this would be a GIF you would add to the README.

If it's the latter, I could pretty easily make a GIF for this using [asciinema](https://asciinema.org/). Just `cat` the unformatted file beforehand, run the formatter binary on the file, and `cat` the new file after it's been changed.

If it's the former, that's a bit more complicated. You might be able to create some script that uses asciinema to automate making a recording, but I'm not sure how that'd work.

There's also [Carbon](https://carbon.now.sh/), but it doesn't appear to support SystemVerilog syntax.

fangism, 2021-02-21:
Hi, the useful artifacts for this project are:

* before/after formatting text as images (e.g. PNG) or vector graphics (SVG) -- these are just still frames for animation (so this is .txt -> .png)
* a script or workflow for assembling those images into an animation (GIF or similar)

It need not be a terminal or session recording (though asciicinema looks cool).

The image could be used in any number of slideshow or video tools, but I figure GIFs are an easy way to start, with with likely open-source image tools that can work offline for the job.

Able to help?

SeanMcLoughlin, 2021-03-02:
Probably not with this issue.

Verible is interesting, though. So I'll look around at other open bugs.