chipsalliance/verible #545: [kythe] linkify anonymous struct/unions member references

fangism, 2020-10-19:
## Input test case

```systemverilog
function f;
  struct { int x; int y; } data;
  int sum = data.x + data.y;
endfunction
```

## Describe what is wrong or missing

`x` in `data.x` should point to the member inside the anonymous struct.  Likewise for `y`.

This is also plausible in ports:

```systemverilog
function f(struct { int x; int y; } data);
  int sum = data.x + data.y;
endfunction
```

MinaToma, 2020-10-21:
Related to #540

MinaToma, 2020-10-22:
Are the missing `()` in the first function intended?

fangism, 2020-10-22:
> Are the missing `()` in the first function intended?

You may add them if you like.  It's syntactically legal either way.

```systemverilog
function int f();
  struct { int x; int y; } data;
  return data.x + data.y;
endfunction
```

commit d2f99829 (2020-11-13): PR #602: Function arguments
Fixes #545
Fixes #540

GitHub PR https://github.com/google/verible/pull/602

Copybara import of the project:

  - da57dfd9389f18634dba4f5721b4364d99d5f507 Function arguments by MinaToma <minatoma1999@gmail.com>
  - e9bdf3ef109b44ea293234bc780873746a27067b remove TODO by MinaToma <minatoma1999@gmail.com>
  - 34d04fafc3b7aa9fe5d089e489fa19f81b2c1502 tests for function port unpacked dimension by MinaToma <minatoma1999@gmail.com>
  - 4dd9f847bf83c858c34c00d4d7b3554da02f1d6e Merge 34d04fafc3b7aa9fe5d089e489fa19f81b2c1502 into 4323a... by Mina Toma <39103197+MinaToma@user [... cut]
closed 2020-11-13 by fangism