chipsalliance/verible #547: [kythe] branch isn't identified as a variable

MinaToma, 2020-10-23:
## Input test case

```systemverilog
class riscv_jump_instr extends riscv_directed_instr_stream;
    riscv_instr          branch;
endclass
```

## Describe what is wrong or missing

CST for this code:
```
Node @3 (tag: kDataDeclaration) {
        Node @1 (tag: kInstantiationBase) {
          Node @0 (tag: kInstantiationType) {
            Node @0 (tag: kReferenceCallBase) {
              Node @0 (tag: kReference) {
                Node @0 (tag: kLocalRoot) {
                  Node @0 (tag: kUnqualifiedId) {
                    Leaf @0 (#SymbolIdentifier @3323-3334: "riscv_instr")
                  }
                }
              }
            }
          }
          Node @1 (tag: kVariableDeclarationAssignmentList) {
            Node @0 (tag: kVariableDeclarationAssignment) {
              Leaf @0 (#"branch" @3344-3350: "branch")
              Node @1 (tag: kUnpackedDimensions) {
              }
            }
          }
        }
        Leaf @2 (#';' @3350-3351: ";")
      }
```

Expected CST:
```
Node @3 (tag: kDataDeclaration) {
        Node @1 (tag: kInstantiationBase) {
          Node @0 (tag: kInstantiationType) {
            Node @0 (tag: kReferenceCallBase) {
              Node @0 (tag: kReference) {
                Node @0 (tag: kLocalRoot) {
                  Node @0 (tag: kUnqualifiedId) {
                    Leaf @0 (#SymbolIdentifier @3323-3334: "riscv_instr")
                  }
                }
              }
            }
          }
          Node @1 (tag: kVariableDeclarationAssignmentList) {
            Node @0 (tag: kVariableDeclarationAssignment) {
              Leaf @0 (#SymbolIdentifier @3344-3350: "branch")
              Node @1 (tag: kUnpackedDimensions) {
              }
            }
          }
        }
        Leaf @2 (#';' @3350-3351: ";")
      }
```

In CST `branch` here should be `SymbolIdentifier` and not `branch`

fangism, 2020-10-23:
Not a kythe issue so much as it is a lexer issue:

https://cs.opensource.google/verible/verible/+/master:verilog/parser/verilog.y;drc=cb7d6286758ebbb446cb9ccb87d5bfff350e94cf;l=533
https://cs.opensource.google/verible/verible/+/master:verilog/parser/verilog.y;drc=cb7d6286758ebbb446cb9ccb87d5bfff350e94cf;l=793

`branch` is a Verilog-AMS keyword, and conservatively, the lexer and parser interpret it as such.  I don't have dialect selection implemented yet.

fangism, 2020-10-23:
Dialect-selection would mean a configurable post-lexer pass that downgrades select keywords into `SymbolIdentifier` before passing it onto the parser.

fangism, 2020-11-10:
Status: Currently, this is worked around in Verible's kythe extractor, the `branch` keyword (when used as an identifier) is ignored safely.

hzeller, 2022-03-01:
Is fixed since https://github.com/chipsalliance/verible/commit/ef935c9036984a4583df5510f813f4e66db0799f in which `branch` is qualified as `SymbolIdentifier` as it is never used in the grammar.