chipsalliance/verible #549: assignment considered as "kDataDeclaration"

MinaToma, 2020-10-23:
In this case:
```
m = 1;
```

The current CST for this code:
```
Parse Tree:
Node @0 (tag: kDescriptionList) {
  Node @0 (tag: kDataDeclaration) {
    Node @0 (tag: kQualifierList) {
    }
    Node @1 (tag: kInstantiationBase) {
      Node @0 (tag: kInstantiationType) {
        Node @0 (tag: kDataType) {
        }
      }
      Node @1 (tag: kVariableDeclarationAssignmentList) {
        Node @0 (tag: kVariableDeclarationAssignment) {
          Node @0 (tag: kUnqualifiedId) {
            Leaf @0 (#SymbolIdentifier @0-1: "m")
          }
          Node @1 (tag: kUnpackedDimensions) {
          }
          Node @2 (tag: kTrailingAssign) {
            Leaf @0 (#'=' @2-3: "=")
            Node @1 (tag: kExpression) {
              Node @0 (tag: kNumber) {
                Leaf @0 (#TK_DecNumber @4-5: "1")
              }
            }
          }
        }
      }
    }
    Leaf @2 (#';' @5-6: ";")
  }
}
```

correct me if I am wrong, but should this be `kDataDeclaration`?

MinaToma, 2020-10-23:
@fangism can you look at this issue and close it if this is intended to be parsed like this.

fangism, 2020-10-23:
So assignments aren't syntactically allowed in the root package scope, however this is parsed as an implicit-typed data declaration with an initialization (global variable).

MinaToma, 2020-10-23:
So this is correct?

MinaToma, 2020-10-23:
This is considered `kVariableDefinition` in Kythe extractor BTW.

fangism, 2020-10-23:
This is correct.

MinaToma, 2020-10-23:
Closing the issue as this was correct.

closed 2020-10-23 by MinaToma