chipsalliance/verible #556: travis-ci: builds time out

fangism, 2020-10-30:
Since kythe was added as a dependency, most travis-builds have been timing out.

First bad:
https://travis-ci.org/github/google/verible/jobs/739104819

Last good:
https://travis-ci.org/github/google/verible/builds/738921915
https://travis-ci.org/github/google/verible/builds/738413728

Things one could look into, to name a few:

* .github/travis/script.sh -- adding some time commands or timestamps
* bazel -j parallelization, already max'd?
* Maintaining a persistent bazel build cache from build-to-build (but also have a way to reset the cache, just-in-case)

fangism, 2020-10-30:
b/171743791

fangism, 2020-10-30:
@mithro anyone on AntMicro team available to help?

fangism, 2020-10-30:
Most recent timeout: https://travis-ci.org/github/google/verible/builds/740056016

mithro, 2020-10-30:
@kgugala - Can you get someone to look at this?

kgugala, 2020-10-30:
@fangism can we migrate the CI to e.g. github actions? there is 6h time limit in actions

fangism, 2020-10-30:
No objection from me.  (I am still interested if there are some easy speed-ups though.)

mglb, 2020-11-03:
I've added timestamps to script.sh. Results: 
* Environment initialization takes < 1 min
* `bazel test` takes about 29 min
* `bazel build` runs until timeout.

See build log for more details: https://travis-ci.com/github/antmicro/verible/jobs/423717516

Running test and build in separate jobs solves the problem. Each job runs in parallel for 29 minutes: https://travis-ci.com/github/antmicro/verible/builds/197663893

fangism, 2020-11-03:
Thanks, I commented in the PR: can we try matching the `-c opt` for both build and test, for the sake of caching?

tgorochowik, 2020-11-10:
@fangism should we still migrate the CI to github actions? It probably makes sense to do that to avoid similar issues in the future.

mithro, 2020-11-10:
@tgorochowik - Yes please move to GitHub Actions ASAP.

fangism, 2020-11-19:
All former Travis-CI actions have been migrated to github actions.

closed 2020-11-19 by fangism