chipsalliance/verible #568: Inconsistency in parsing pure virtual function

MinaToma, 2020-11-04:
**Describe the bug**

Function name identifier is wrapped inside `kDataTypeImplicitBasicIdDimensions`.
**To Reproduce**

```systemverilog
virtual class BaseClass;
   extern function void f(int x);
   pure virtual function int getData();
endclass
```
**Actual behavior:**

Current CST:
```
Node @1 (tag: kForwardDeclaration) {
        Node @0 (tag: kQualifierList) {
          Leaf @0 (#"extern" @41-47: "extern")
        }
        Node @1 (tag: kFunctionPrototype) {
          Leaf @1 (#"function" @48-56: "function")
          Node @3 (tag: kDataTypeImplicitBasicIdDimensions) {
            Node @0 (tag: kDataType) {
              Leaf @1 (#"void" @57-61: "void")
            }
            Node @1 (tag: kUnqualifiedId) {
              Leaf @0 (#SymbolIdentifier @62-63: "f")
            }
            Node @2 (tag: kUnpackedDimensions) {
            }
          }
          Node @4 (tag: kParenGroup) {
            Leaf @0 (#'(' @63-64: "(")
            Node @1 (tag: kPortList) {
              Node @0 (tag: kPortItem) {
                Node @1 (tag: kDataTypeImplicitBasicIdDimensions) {
                  Node @0 (tag: kDataType) {
                    Node @1 (tag: kDataTypePrimitive) {
                      Leaf @0 (#"int" @64-67: "int")
                    }
                    Node @3 (tag: kPackedDimensions) {
                    }
                  }
                  Node @1 (tag: kUnqualifiedId) {
                    Leaf @0 (#SymbolIdentifier @68-69: "x")
                  }
                  Node @2 (tag: kUnpackedDimensions) {
                  }
                }
              }
            }
            Leaf @2 (#')' @69-70: ")")
          }
          Leaf @5 (#';' @70-71: ";")
        }
      }
      Node @2 (tag: kForwardDeclaration) {
        Node @0 (tag: kQualifierList) {
          Leaf @0 (#"pure" @75-79: "pure")
          Leaf @1 (#"virtual" @80-87: "virtual")
        }
        Node @1 (tag: kFunctionPrototype) {
          Leaf @1 (#"function" @88-96: "function")
          Node @3 (tag: kDataTypeImplicitBasicIdDimensions) {
            Node @0 (tag: kDataType) {
              Node @1 (tag: kDataTypePrimitive) {
                Leaf @0 (#"int" @97-100: "int")
              }
              Node @3 (tag: kPackedDimensions) {
              }
            }
            Node @1 (tag: kUnqualifiedId) {
              Leaf @0 (#SymbolIdentifier @101-108: "getData")
            }
            Node @2 (tag: kUnpackedDimensions) {
            }
          }
          Node @4 (tag: kParenGroup) {
            Leaf @0 (#'(' @108-109: "(")
            Leaf @2 (#')' @109-110: ")")
          }
          Leaf @5 (#';' @110-111: ";")
        }
      }
```
**Expected behavior**

Add function header and re-shape the CST correctly for `pure function` and `extern function`.

fangism, 2020-11-04:
Confirmed, I need to unpack `kDataTypeImplicitBasicIdDimensions` and re-pack it for consistency.

MinaToma, 2020-11-05:
There is still some inconsistency. 

CST for headers in pure functions:
```
Node @1 (tag: kFunctionPrototype) {
  Node @0 (tag: kFunctionHeader) {
    Leaf @1 (#"function" @35-43: "function")
    Node @3 (tag: kUntagged) {
      Node @0 (tag: kDataType) {
        Leaf @1 (#"void" @44-48: "void")
      }
      Node @1 (tag: kUnqualifiedId) {
        Leaf @0 (#SymbolIdentifier @49-50: "f")
      }
    }
    Node @4 (tag: kParenGroup) {
      Leaf @0 (#'(' @50-51: "(")
      Leaf @2 (#')' @56-57: ")")
    }
    Leaf @5 (#';' @57-58: ";")
  }
}
```

CST for headers normal function:
```
Node @0 (tag: kFunctionHeader) {
  Node @0 (tag: kQualifierList) {
    Leaf @0 (#"virtual" @149-156: "virtual")
  }
  Leaf @1 (#"function" @157-165: "function")
  Node @3 (tag: kDataType) {
    Node @1 (tag: kDataTypePrimitive) {
      Leaf @0 (#"int" @166-169: "int")
    }
    Node @3 (tag: kPackedDimensions) {
    }
  }
  Node @4 (tag: kUnqualifiedId) {
    Leaf @0 (#SymbolIdentifier @170-177: "getData")
  }
  Node @5 (tag: kParenGroup) {
    Leaf @0 (#'(' @177-178: "(")
    Leaf @2 (#')' @178-179: ")")
  }
  Leaf @6 (#';' @179-180: ";")
}
```

`kParenGroup` at index `4` in `pure function header` and at index `5` in `normal function header`.
function identifier in `pure function` is within `kUntagged` node and not the header directly at index `4` like `normal function header`

commit f9d44276 (2020-11-05): [CST] Shape FunctionPrototypes with FunctionHeader nodes for consistency.

FunctionHeader nodes are common to both prototypes and full declarations.

Update many unit-tests to check for precise token substring ranges.

fixes #568

PiperOrigin-RevId: 340816742
commit 49b18808 (2020-11-05): [CST] Give function prototype headers the same structure as declaration counterparts.

fixes #568

PiperOrigin-RevId: 340909207
closed 2020-11-05 by fangism