chipsalliance/verible #569: Inconsistency in pure task

MinaToma, 2020-11-04:
**Describe the bug**

The CST of `pure tasks` should contain a `kTaskHeader` node.

**To Reproduce**

```systemverilog
class Base2;
   pure virtual task disp();
endclass
```

**Actual behavior:**
Current CST:
```
Node @1 (tag: kTaskPrototype) {
          Leaf @1 (#"task" @407-411: "task")
          Leaf @3 (#SymbolIdentifier @412-416: "disp")
          Node @4 (tag: kParenGroup) {
            Leaf @0 (#'(' @416-417: "(")
            Leaf @2 (#')' @417-418: ")")
          }
          Leaf @5 (#';' @418-419: ";")
        }
```

**Expected behavior**

Re-shape CST for pure tasks to have `kTaskHeader`.

fangism, 2020-11-04:
Confirmed, thanks.

MinaToma, 2020-11-05:
There is some inconsistency with normal task header:

CST for pure task header:
``
Node @1 (tag: kTaskPrototype) {
  Node @0 (tag: kTaskHeader) {
    Leaf @1 (#"task" @394-398: "task")
    Leaf @3 (#SymbolIdentifier @399-403: "disp")
    Node @4 (tag: kParenGroup) {
      Leaf @0 (#'(' @403-404: "(")
      Leaf @2 (#')' @404-405: ")")
    }
    Leaf @5 (#';' @405-406: ";")
  }
}
```

CST for tasks:
```
Node @0 (tag: kTaskHeader) {
  Node @0 (tag: kQualifierList) {
    Leaf @0 (#"virtual" @434-441: "virtual")
  }
  Leaf @1 (#"task" @442-446: "task")
  Node @3 (tag: kUnqualifiedId) {
    Leaf @0 (#SymbolIdentifier @447-451: "disp")
  }
  Node @4 (tag: kParenGroup) {
    Leaf @0 (#'(' @451-452: "(")
    Leaf @2 (#')' @452-453: ")")
  }
  Leaf @5 (#';' @453-454: ";")
}
```

`SymbolIdentifier` in pure task isn't wrapped inside a `kUnqualifiedId` node.

fangism, 2020-11-05:
Fix underway.

commit 4608bbc2 (2020-11-05): Provide TaskHeader CST node for TaskPrototypes.

This improves structural consistency, and allows reuse of TaskHeader accessor
functions.

Update CST/tasks_test.cc matching unit-tests.

fixes #569

PiperOrigin-RevId: 340711308
commit 773fa5e9 (2020-11-05): [CST] Make task headers have same struct in protos and decls.

fixes #569

PiperOrigin-RevId: 340923969
closed 2020-11-05 by fangism