chipsalliance/verible #571: Crash while parsing "E(node.Tag().tag) == expected_node_enum (kFunctionCall vs. kReference)"

MinaToma, 2020-11-04:
**Describe the bug**

`verilble-verilog-syntax` crash on these two files and give `"E(node.Tag().tag) == expected_node_enum (kFunctionCall vs. kReference)`.

**To Reproduce**

run `verible-verilog-syntax` on:
- opentitan/hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv
- uvm/distrib/src/deprecated/uvm_resource_converter.svh

Try the below case:

```systemverilog
class m_uvm_resource_sprint_converter#(type T=int) extends m_uvm_resource_converter#(T);
   local static m_uvm_resource_sprint_converter #(T) m_singleton;

   virtual function string convert2string(T val);
      return $sformatf("(%s) %0s", uvm_type_utils#(T)::typename(val),
                       (val == null) ? "(null)" : {"\n",val.sprint()});
   endfunction
   
   `_local function new();
   endfunction

   // Function- register
   // Register this policy class as the resource value conversion function
   // for this resource type.
   //
   //| void'(uvm_resource_sprint_converter#(my_obj)::register());
   //
   static function bit register();
      if (m_singleton == null) m_singleton = new();
      uvm_resource#(T)::m_set_converter(m_singleton);
      return 1;
   endfunction
endclass
```

fangism, 2020-11-04:
Smaller test case:

```systemverilog
class c;
   `_local function new();
   endfunction

   static function bit r();
      if (m == null) m = new();
      uvm_resource#(T)::m_set_converter(m_singleton);
   endfunction
endclass
```

So far I can see that this is tripping syntax error recovery due to the `` `_local `` token.

fangism, 2020-11-04:
Fix coming: recovering syntax error up to the first `endfunction`.

fangism, 2020-11-04:
The fix I have so far is good for the small test case, but the OpenTitan file:

https://github.com/lowRISC/opentitan/blob/master/hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv

looks un-parse-able in many ways due to preprocessing macros.  (We don't have a preprocessor yet.)
I will see if I can prevent the crash nevertheless.

commit 98de674c (2020-11-05): [syntax error-recovery] recover around covergroups and bin/option items.

Now the following sources only fail with syntax errors *without* crashing:

https://github.com/lowRISC/opentitan/blob/master/hw/vendor/lowrisc_ibex/vendor/google_riscv-dv/src/riscv_instr_cover_group.sv https://github.com/accellera/uvm/blob/master/distrib/src/deprecated/uvm_resource_converter.svh

fixes #571

PiperOrigin-RevId: 340763894
closed 2020-11-05 by fangism