chipsalliance/verible #574: [kythe] automatic inter-file dependency detection

fangism, 2020-11-06:
Currently, the verilog-kythe extractor requires an input file list that is already dependency sorted, with symbol definitions always in earlier files than references.  This places the burden on the user to pass in such an ordered file list.  (The consequence of incorrect ordering now is just missing def/ref edges, but aside from that things should still work.).  Since we have the ability to construct partial symbol tables and detect (punt) undefined references, it should be possible to establish dependency edges between files based on root-level symbol definitions and references.

The dependency graph should not only be used internally for project fact/edge extraction, but it could be output as intermediate information for other tools to consume, e.g. unordered file-list in, ordered file-list out.

If you output lines of `a b` where `b` depends on `a` (an edge), you could even pipe that through UNIX `tsort` (topological sort).

fangism, 2020-11-06:
blocked by #563

fangism, 2020-11-06:
This dependency information could also be used to optimize extraction in that only anti-dependencies of modified files have to be re-extracted (incremental extraction).  For low-latency on-the-fly re-indexing!

suzizecat, 2021-04-09:
Hi ! As I see in your commit 0e00783 , I understand that `verible-verilog-project file-deps` should be used to scan for dependencies. 
Could you confirm that ? 

Also, I seem to have an issue with it on a pretty big testcase. Is there known issues preventing `file-deps` from outputing any kind of dependency information ? (I got a lot of stuff on stderr, but nothing on stdout)

fangism, 2021-04-09:
Hello, that is the intent, though it may not be complete.  This started out as a proof-of-concept, you might be among the first to try it on larger projects and find early bugs.  This mini-project would benefit greatly from building up a suite of small demonstrative test cases, working our way up to more intricate ones.  Can you share any test cases that demonstrate problems, preferably reduced?

suzizecat, 2021-04-09:
Hi,
Unfortunately, I cannot share this testcase, and the others testcase I might have are not big enough to stress anything.

If I stumble upon this issue on a testcase I can share, I'll provide it.
One potential problem could be from symbols defined multiple times, I'll test that out.

Just two points :
- Could it be possible to output the dependency graph through a Kythe schema (as Kythe propose the depends edge) ?
- With this question comes another : could it be possible to integrate this in the `kythe-extractor` to get  file ordering (if I understand correctly, that was the original purpose of this mini-project) and the indexer run on the same go, hence being probably much more efficient than running the dependency detection and then the index ?