chipsalliance/verible #581: Support for SV nettype

msfschaffner, 2020-11-06:
**Summary**

The SV `nettype` keyword is currently not supported by Verible.
Since analog circuit models may sometimes contain this keyword, it would be nice if this was supported as well.

**Test cases and examples**

The more examples you provide, the easier it will be for someone to come along
and implement it.

```systemverilog
nettype real awire;
```

**Additional context**

See also https://verificationacademy.com/forums/systemverilog/difference-between-nettype-and-wire-logic for examples and more references.

msfschaffner, 2020-11-06:
b/172677968

msfschaffner, 2020-11-06:
CC @arnonsha

msfschaffner, 2020-11-06:
This came up in https://github.com/lowRISC/opentitan/pull/3952#discussion_r517198761

fangism, 2020-11-08:
This is LRM-2017, section 6.6.7 User-defined nettypes.

commit 84495fd0 (2020-11-09): Accept net_type_declarations syntax.

e.g. nettype real awire;

Also format these on their own lines.

fixes #581

PiperOrigin-RevId: 341447474
closed 2020-11-09 by fangism