chipsalliance/verible #582: Feature request - add documentation extractor

svenka3, 2020-11-08:
Hello,
  Based on a discussion on Issue - https://github.com/google/verible/issues/340 

I would like to see a documentation extraction feature/tool.  For TB in SV/UVM, think of something similar to Doxygen/NaturalDocs for reference outputs. Adding UML diagrams, inheritance trees would be very handy - consider exporting PNG/SVG etc. for end-user document integration. 

Thanks
Srini

fangism, 2020-11-09:
Thanks for filing this idea!  Some thoughts:

* We should be able to convert comments into documentation snippets, provided we establish how to "attach" comments to non-comments.  e.g. IIRC with Doxygen `///` comments apply to the next thing, and `///<` comments apply to the thing on the left.
* A lot of documentation can be generated (modules, ports, interfaces, types, inheritance) based on the existing concrete syntax tree, and essentially information that the Verible+Kythe project already.  We should be leveraging that as much as possible (some refactoring needed).  @ivan444 any thoughts around standard practices of extracting comments into hover-text or additional metadata? or any prior art of using Kythe to generate documentation?

ivan444, 2020-11-09:
> any thoughts around standard practices of extracting comments into hover-text or additional metadata? 

Kythe has support for extracting documentation. I didn't explore it, but I guess you need a tool like Doxygen to provide you the annotations which you convert to Kythe facts.

> or any prior art of using Kythe to generate documentation?

I'm not aware of it. But playing with https://github.com/cayleygraph/cayley on top of the extracted Kythe facts might get you somewhere.