chipsalliance/verible #584: [travis-CI] failing git-init on kythe repo

fangism, 2020-11-09:
After fixing #556, the next issue is related to Kythe as a dependency.

See this log:
https://travis-ci.org/github/google/verible/jobs/742535455

```
INFO: Repository io_kythe instantiated at:
  /src/verible/verible-v0.0-759-g7941857/WORKSPACE:143:15: in <toplevel>
Repository rule git_repository defined at:
  /root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/bazel_tools/tools/build_defs/repo/git.bzl:199:33: in <toplevel>
ERROR: An error occurred during the fetch of repository 'io_kythe':
   Traceback (most recent call last):
	File "/root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/bazel_tools/tools/build_defs/repo/git.bzl", line 181, column 30, in _git_repository_implementation
		update = _clone_or_update(ctx)
	File "/root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/bazel_tools/tools/build_defs/repo/git.bzl", line 36, column 20, in _clone_or_update
		git_ = git_repo(ctx, directory)
	File "/root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 91, column 12, in git_repo
		_update(ctx, git_repo)
	File "/root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 101, column 9, in _update
		init(ctx, git_repo)
	File "/root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 118, column 15, in init
		_error(ctx.name, cl, st.stderr)
	File "/root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 190, column 9, in _error
		fail("error running '%s' while working with @%s:\n%s" % (command_text, name, stderr))
Error in fail: error running 'git init /root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/io_kythe' while working with @io_kythe:
src/main/tools/process-wrapper-legacy.cc:80: "execvp(git, ...)": No such file or directory
ERROR: no such package '@io_kythe//': error running 'git init /root/.cache/bazel/_bazel_root/2ccca2c4c14abc5021f2a7b18d0dec9b/external/io_kythe' while working with @io_kythe:
src/main/tools/process-wrapper-legacy.cc:80: "execvp(git, ...)": No such file or directory
INFO: Elapsed time: 7.206s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded)
FAILED: Build did NOT complete successfully (0 packages loaded)
The command '/bin/sh -c bazel build --workspace_status_command=bazel/build-version.sh -c opt //...' returned a non-zero code: 1
The command "./.github/travis/script.sh" exited with 1.
```

FYI @ivan444 .

fangism, 2020-11-17:
Could this be a matter of making sure `git` is installed on the executor?  (Bazel is invoking git as a means of fetching a repo.)

ivan444, 2020-11-17:
Most likely. Kythe is introduced as git_repo, not http_repo.

fangism, 2020-11-17:
@kgugala @wtatarski Have some bandwidth to investigate?

wtatarski, 2020-11-17:
@fangism I installed some lacking dependencies but now I see that some libs are outdated:
https://github.com/antmicro/verible/runs/1413572108?check_suite_focus=true
So I guess here is some work to do.

fangism, 2020-11-17:
Other test scripts we have use bazel 3.7.0.

https://cs.opensource.google/search?q=BAZEL_VERSION&sq=&ss=verible%2Fverible

ivan444, 2020-11-17:
Maybe a simpler approach would be to integrate Kythe in a different way. We
don't need all of it.
But I don't have a concrete proposal at the moment.

mithro, 2020-11-17:
Doesn't #603 obsolete this issue?

mithro, 2020-11-17:
IE - As #603 is merged, Travis-CI is no longer being run.

fangism, 2020-11-17:
This same problem occurs on github action, so I'll update the title.

fangism, 2020-11-17:
Example recent log:
https://github.com/antmicro/verible/runs/1411135012?check_suite_focus=true

wtatarski, 2021-06-14:
Fixed in #608

closed 2021-06-14 by wtatarski